qmk/qmk_firmware #5780: Space Cadet shift starts with extra Shift up and down, and frequently double-taps

petergaultney, 2019-05-04:
## Describe the Bug

At my last compilation in January, Space Cadet Shift worked perfectly. The event output from my keyboard for a single Space Cadet shift tap looked like this:

[image]

Now event output for a single keystroke (my tapping term is still 200) looks like this:
[image]

Note the extra Shift down/up before even starting the combination, and note that it then repeats itself twice despite there being only a single tap.

## System Information

 - Keyboard:  Ergodox EZ
   - Revision (if applicable):  
 - Operating system: Mac 10.14
 - AVR GCC version: 4.9.2
 - ARM GCC version: (unknown, using Ergodox configurator and compiler)

 - QMK Firmware version: 0.6.325
 - Any keyboard related software installed? 
   - [*] Karabiner - I have disabled all Space Cadet-related things in Karabiner

XScorpion2, 2019-05-04:
Hmm, ya, I can add a check if the tap & hold mods are the same to skip the extra up and down in this case. The double tap might be due to debounce, perf has been increased quite a bit and @drashna was saying he needed to increase debounce time because of this.

XScorpion2, 2019-05-04:
Here ya go: https://github.com/qmk/qmk_firmware/pull/5781

closed 2019-05-05 by drashna

skorokithakis, 2020-03-09:
@petergaultney Which application did you use to generate those keystroke logs?

petergaultney, 2020-03-09:
Karabiner Elements on Mac. https://pqrs.org/osx/karabiner/