qmk/qmk_firmware #5792: i2c_master library licensing unclear

yurikhan, 2019-05-05:
The drivers/avr/i2c_{master,slave}.* libraries have an authorship notice at the top of both files and links to the upstream repository:

```
/* Library made by: g4lvanix
 * Github repository: https://github.com/g4lvanix/I2C-master-lib
 */
```

[In that repo][0], there is no LICENSE file, nor are licensing terms stated in the README.md, and there is a (neglected) issue g4lvanix/I2C-master-lib#2 asking for a license.

[0]: https://github.com/g4lvanix/I2C-master-lib/

Before that, QMK used to contain TWIlib by Chris Herring, which also linked to the [original web site][1] which does not contain any explicit license.

[1]: http://www.chrisherring.net/all/tutorial-interrupt-driven-twi-interface-for-avr-part1/

What is the policy in QMK for including libraries that are formally non-free?

skullydazed, 2019-05-08:
Thanks for bringing this to our attention. We've been examining the situation.

It looks like there is an equivalent replacement library available here: http://homepage.hispeed.ch/peterfleury/avr-software.html

While Peter Fleury doesn't follow best practices for documenting the license (for example, no preamble, no license text) he makes it clear on the webpage you download from that it is distributed under the GPL. In the source download itself he further clarifies GPLv3 in `manual__i2cmaster.html`:

```
Author
    Peter Fleury pfleury@gmx.ch http://tinyurl.com/peterfleury
Copyright
    (C) 2015 Peter Fleury, GNU General Public License Version 3
```

skullydazed, 2019-05-08:
(sorry for the quick comment in succession, clicked comment too soon.)

As a path forward I or someone else will open a PR to replace those files, and we'll solicit tests from as many people as we can. It will take a take or two before I have time to do that.

yurikhan, 2019-05-08:
That looks good, but seems to be master side only. QMK’s i2c_slave is also by g4lvanix and formally non-free.

However maybe we should ask g4lvanix if he is willing to release his I2C libraries under a license compatible with GPL?

pelrun, 2019-05-08:
Minor point of contention - it's not "formally" non-free, which would require an explicit license statement to that effect. Anything like that would never have been allowed into QMK in the first place (hence why we don't support Nordic chipsets...)

skullydazed, 2019-05-08:
We have reached out to @g4lvanix, but in the meantime we are working on alternatives. We will need to find a good story for slave support, which may mean a new implementation.

Rather than get hung up on whether something is formally non-free or not, we prefer to stick to more concrete terms. Any code that is not under the GPL needs to be under a GPL-compatible license. This code being merged was an oversight on our part.

drashna, 2019-07-15:
This has been updated, and the code is officially GPL3 licenses.   

https://github.com/g4lvanix/I2C-master-lib/pull/4#issuecomment-511208652

drashna, 2019-07-15:
This is fixed, now!

closed 2019-07-15 by drashna