qmk/qmk_firmware #5804: document bug of contributing.md

ymzcdg, 2019-05-06:
## Describe the Bug
document issue in contributing.md 
1.there is a sentence"Keyboards are the raison d'être for QMK. " What does "d'être"mean?
2."you can click "Suggest An Edit" at the top of each page on http://docs.qmk.fm/." I can't find.
3.I have a question :what does the "squashed"mean in "All Keymap PR's are squashed, so if you care about how your commits are squashed you should do it yourself" thanks.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2019-05-06:
1. [raison d'être](https://www.merriam-webster.com/dictionary/raison%20d'%C3%AAtre) - "reason or justification for existence"
2. I think this is a holdover from the old document site that we used to use.  It should be fixed, yeah. 
3. This is a git merge thing.  There are 3 types for merges, actually: 
   1. (straight) Merge. This adds a "merge" commit, and adds the commits from your branch/PR to the repo. This maintains the commit IDs, and other info.  But isn't as "clean"
    2. Squash merge. This "squashes" the commits down to a single commit, and then adds that single commit to the repo.  This is very clean, but loses any commit IDs, and signing status
    3. Rebase.  This adds all of the commits from the PR, but gives them a new commit ID, and removes any signing of the commits. 
    
    Depending on the type of PR, and the number of commits (and who is merging it) will change how the PR is merged into the repo.   

But maybe we should add a link to GitHub's page on this: https://help.github.com/en/articles/merging-a-pull-request#merging-a-pull-request-on-github

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc