qmk/qmk_firmware #5808: Ergodash RGB broken on slave half (left) broken by recent PR

yourfate, 2019-05-07:
I use an ergodash, the changes made in this PR: https://github.com/qmk/qmk_firmware/pull/5772 broke RGB on the slave-half of my keyboard. When I turn on RGB it only turns on on the master half (right).

RGB worked fine on both halves of the keyboard before this PR was merged, a colleague of mine with the same setup (right hand master) is experiencing the same problem.

I'm using this keymap / config.h / rules.mk: https://gitlab.com/youRFate/keymaps/tree/master/ergodash/yf
I built the firmware by copying the `yf` folder into `keyboards/ergodash/rev1/keymaps`, then using `make ergodash/rev1:yf`. The resulting .hex file can be found here: https://gitlab.com/youRFate/keymaps/blob/master/ergodash/ergodash_rev1_yf.hex
I flashed the new firmware on both halves of the keyboard. I'm using elite-c controllers.

The solder bridges for master/slave can be seen here: https://images.yourfate.org/uploads/big/07eb2d2c298519841c2cbce5aac1d6ef.jpg

## System Information

 - Keyboard:  Ergodash
   - Revision:  1.2 (rev1)
 - Operating system: Ubuntu 18.04
 - AVR GCC version: 5.4.0
 - ARM GCC version: (15:6.3.1+svn253039-1build1) 6.3.1 20170620
 - QMK Firmware version: 0.6.364

omkbd, 2019-05-07:
The last section of the build guide describes the solution to this problem. Currently only Japanese is written, but I am translating into English.
https://github.com/omkbd/ErgoDash/blob/master/Doc/build.md#12-firmware%E3%81%AE%E6%9B%B8%E3%81%8D%E8%BE%BC%E3%81%BF

yourfate, 2019-05-07:
Using google translate I manged to find out what I had to do. I chose to use `#undef` in the config.h inside my keymap instead of commenting it out in the general config.h so I can leave QMK unchanged. It works now: https://gitlab.com/youRFate/keymaps/blob/master/ergodash/yf/config.h#L36.

Thank you!

closed 2019-05-07 by yourfate