qmk/qmk_firmware #5819: Mod Tap or double function keys are sticking in BIOS

souew, 2019-05-08:
I have a mod tap keys on a home row like this:
`LSFT_T(KC_A), LGUI_T(KC_S), LCTL_T(KC_D), LALT_T(KC_F)`
And also a double function split spacebar like this:
`LT(1, KC_SPC), LT(2, KC_ENT)`
It works in the OS, but in BIOS these keys are sticking, and pressing themselves on their own after my initial press, normal keys with only one function doesn't behave in this way, also if I press any other key immediately after pressing the double function key it doesn't stick, even if the pressed key also has double function on it.
My TAPPING_TERM is 150, tried 500, didn't help.
My DEBOUNCING_DELAY is 5, tried 0, didn't help.
I don't have anything set about NKRO in config.h
Tried different combinations of
```
#define IGNORE_MOD_TAP_INTERRUPT
#define PERMISSIVE_HOLD
#define TAPPING_FORCE_HOLD
#define RETRO_TAPPING
#define PREVENT_STUCK_MODIFIERS
```
or all of them, or none of them, didn't help.
My hardware is custom Planck-like build with atmega32u4.

souew, 2019-05-08:
Slight addition, if I tap another key immediately and that key also has double function on it, it also sticks, but if I hold it and release, it doesn't stick.
Tried #undef IGNORE_MOD_TAP_INTERRUPT it did nothing in BIOS, but worsened the OS behavior, mod keys sometimes were not registering at all.

drashna, 2019-05-08:
It sounds like the keyboard may not be properly initializing, due to the BIOS. 

If you're compiling from source, try adding this to your `config.h` and see if that helps:
```c
#define NO_USB_STARTUP_CHECK
```

souew, 2019-05-08:
Unfortunately it didn't help. I tested it on a more modern PC with UEFI BIOS and it works fine there. My motherboard is Gigabyte x58a-ud7 (LGA1366), and it has most up to date BIOS flashed. Maybe it's something with the refresh or pooling rate of the keyboard, the BIOS itself feels sluggish. Tried different USB ports also.
Anything else I can try? It's quite important to me as I need to type the GRUB decryption passphrase.

drashna, 2019-05-08:
Well, which keyboard is this? 

Also, while you may not have NKRO enabled, it .... may still be enabled. 

It may be worth resetting the EEPROM, to ensure that NKRO is not actually enabled, and that just 6KRO is enabled (the default mode)

souew, 2019-05-08:
Yes I did the reset. The keyboard is a custom handwired build on Teensy 2.0

drashna, 2019-05-08:
Hmm, okay. I'm not sure.

souew, 2019-05-09:
Can I programmically switch layers? I thought maybe I would make layer 0 without double function keys, to type the passphrase in GRUB, and then in bspwmrc I could put some script which will switch me to layer 1 automatically? I run arch linux.

souew, 2019-05-09:
Wait a minute, it's probably better and easier to switch to that special layer in bios to type the passphrase, I did that and it seem to work. So I found a solution, even if it's not optimal, I will close the issue, as I haven't seen anything similar in my extensive searching, I think it's my BIOS/hardware problem.

closed 2019-05-09 by souew