qmk/qmk_firmware #5833: Backlight and underglow LEDs stay on

thepinkturtle, 2019-05-09:
With underglow and backlight, the LEDs on the slave side won't turn completely off. The LEDs on the slave side will dim to a certain point and then stay on and won't get any lower light output.

## Describe the Bug
When the keyboard has underglow and backlight the backlight LEDs on the slave side don't turn off and won't dim completely. The LEDs will dim to a certain point and then stop, while the master side LEDs will keep getting lower.

## System Information

 - Levinson:  
   - Revision 2  
 - Operating system:
 - Ubuntu 14

 - Will update everything below soon
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

## Additional Context

Lenbok, 2019-05-10:
What version of the code are you running? #3886 was merged just 3 days ago which may fix what you are seeing, so please check.

thepinkturtle, 2019-05-13:
Alright, I'll try it tonight. Hopefully, it fixes it. 😊

thepinkturtle, 2019-05-16:
im using the latest qmk_firmware, i just did a pull today.

thepinkturtle, 2019-11-26:
For the record this was never solved

drashna, 2019-12-01:
Try adding: 
```c
void suspend_power_down_user(void) {
    rgblight_disable_noeeprom();
    backlight_disable();
}

void suspend_wakeup_init_user(void) {
    rgblight_enable_noeeprom();
    backlight_enable();
}
```

This will hard code to turn on... but ... may be a good test.

TD22057, 2022-06-17:
Still a problem.

closed 2022-08-10 by thepinkturtle