qmk/qmk_firmware #5835: Niu Mini uses dfu bootloader, the suggested command is wrong

ivomac, 2019-05-10:
[Here](https://github.com/qmk/qmk_firmware/tree/master/keyboards/niu_mini) it is suggested to flash the Niu Mini with
  
`make niu_mini:default:avrdude`

but it should say

`make niu_mini:default:dfu`

newtonapple, 2019-05-16:
I have the niu_mini as well and I can confirm it should be dfu and not avrdude.  I can send a PR if you want.

ivomac, 2019-05-16:
Sure, I didn't look into how to do that yet. Thanks

drashna, 2019-05-26:
@ivomac You should be able to "edit" the readme fine in the repo.  that should allow you to create a branch in your repo for it, and to open a PR to fix it.

hadibus, 2019-06-07:
@ivomac I ran into the same issue when trying to flash my niu_mini. Hope you don't mind me doing a PR.

PR #6092 (2019-06-09): niu_mini uses dfu bootloader rather than the afrdude bootloader
	modified: readme.md

## Description
Small change to the readme. The afrdude bootloader doesn't work for the niu_mini. The dfu bootloader does work for the niu_mini.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

*  Fixes #5835 

## Checklist

- [x] My code follows the code style of this project.
- [x] My change requires a change to the doc [... cut]
closed 2019-06-09 by drashna