qmk/qmk_firmware #5837: RGB_MATRIX_TYPING_HEATMAP and underglow

jpetermans, 2019-05-10:
The new typing heatmap effect is awesome! However when in this mode on the ALT, underglow processing seems to be completely turned off.  I looked through `typing_heatmap_anim.h` but didn't see anything obvious that would skip leds with underglow flags.

As an example, with other effects and underglow off, using `rgb_matrix_indicators_user` if I temporarily turn on an underglow led(s) as a layer indicator then move off that layer, the led turns off on its own, I don't explicitly have to tell it to turn off. I assume that is being handled by `rgb_matrix.c`.  However, with `RGB_MATRIX_TYPING_HEATMAP` enabled, this does not happen. If I turn on any underglow led, it stays on until I explicitly turn it off.

**So my question:** What is different about how `RGB_MATRIX_TYPING_HEATMAP` sets/handles underglow leds compared to other effects? Maybe this is related to the wider setting `RGB_MATRIX_FRAMEBUFFER_EFFECTS` but I haven't been able to also test `RGB_MATRIX_DIGITAL_RAIN` since it's broken until PR #5811 merges.

Secondary question: Could this effect be combined with #5764 to have both keys and underglow respond to speed?

XScorpion2, 2019-05-10:
Digital rain and Typing heat map both exclusively work on LEDs that are tied to the key matrix. This is mainly due to rgb matrix originally not having underglow support. I have a few ideas on how this could be improved that I want to test out. Not sure if the ideas I have will work and be performant.

As for velocikey support, I do plan on integrating that as well in a future pr.

jpetermans, 2019-05-10:
Do these effects change normal led processing flow when enabled; skipping underglow?  I have a vague sense of how things flow through `rgb_matrix.c`, but not a good one.  The other KEYPRESS/KEYRELEASE don't have the same behavior.

XScorpion2, 2019-05-10:
It's just the way they were designed, they both iterate over the matrix row / col and set colors for those led indexes only.

jpetermans, 2019-05-10:
Thanks for your responses and the heatmap effect! I'm slowing soaking in the rgb matrix flow.

closed 2019-05-10 by jpetermans

BryanSchuetz, 2020-08-11:
Curious if anyone ever found a way to extend heatmap animation to the the underglow.

stolmen, 2024-11-09:
> Curious if anyone ever found a way to extend heatmap animation to the the underglow.

No one will probably see this but hoping it may help someone. https://github.com/stolmen/miryoku_qmk/pull/1/commits/cd175391eff7e3402662875d39b084dba4dad651 this is my quick and dirty implementation which seems to work on my corne.

stolmen, 2025-05-03:
> I see this! I was able to create a custom matrix effect based on your idea
> without needing to modify the QMK core files. Reference the
> rgb_matrix_user.inc file for the code. I am not a good developer, so I'm
> sure you could make something more impressive out of this.
> keyboards/crkbd/keymaps/typer

Awesome glad this helped someone. 😀