qmk/qmk_firmware #5838: keyboards / doro67 / multi / multi.c - Caps Lock LED support

cerement, 2019-05-10:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Caps Lock LED support missing in keyboards / doro67 / multi / multi.c
https://github.com/qmk/qmk_firmware/blob/master/keyboards/doro67/multi/multi.c

Just needs to be copied from keyboards / doro67 / regular / regular.c
https://github.com/qmk/qmk_firmware/blob/master/keyboards/doro67/regular/regular.c
or keyboards / doro67 / rgb / rgb.c
https://github.com/qmk/qmk_firmware/blob/master/keyboards/doro67/rgb/rgb.c

```

/* Copyright 2019 ShadeDream
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */
#include "multi.h"

void matrix_init_kb(void) {
	// put your keyboard start-up code here
	// runs once when the firmware starts up
	
	setPinOutput(E6);
	matrix_init_user();
}

void matrix_scan_kb(void) {
	// put your looping keyboard code here
	// runs every cycle (a lot)

	matrix_scan_user();
}

bool process_record_kb(uint16_t keycode, keyrecord_t *record) {
	// put your per-action keyboard code here
	// runs for every action, just before processing by the firmware

	return process_record_user(keycode, record);
}

void led_set_kb(uint8_t usb_led) {
	// put your keyboard LED indicator (ex: Caps Lock LED) toggling code here

	if (IS_LED_ON(usb_led, USB_LED_CAPS_LOCK)) {
		writePinLow(E6);
	} else {
		writePinHigh(E6);
	}

	led_set_user(usb_led);
}

```

mechmerlin, 2019-05-13:
@ShadeDream can you take care of this?

ShadeDream, 2019-05-14:
@mechmerlin Yep I can take care of it.

PR #5875 (2019-05-15): Added capslock LED support to the Doro67 Multi PCB.
## Description

Added capslock LED support to the Doro67 Multi PCB.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #5838

## Checklist

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIBUTING** document [... cut]
closed 2019-05-15 by drashna