qmk/qmk_firmware #5845: Vagrant: provisioning fails with `/bin/sh: 1: /vagrant/util/qmk_install.sh: not found`

igorrafael, 2019-05-11:
## Describe the Bug

It seems that changing from Ubuntu to Debian (#4905) broke the Vagrant setup. When running `vagrant up`, it fails with the following error: `qmk_firmware: /bin/sh: 1: /vagrant/util/qmk_install.sh: not found`

The box seems to work fine by reverting to the previous commit that affected this file (#4905).

## Additional Context

Commands I ran in PowerShell after confirming that it worked in 37b042a:

```yaml
PS D:\git\qmk> del .\.vagrant\
PS D:\git\qmk> git checkout d8df01ca5e43b6322ba2ef5c7613a871ce69282c
Checking out files: 100% (3779/3779), done.
Previous HEAD position was 37b042a59 Initial fixes for vagrant (#4900)
HEAD is now at d8df01ca5 Align docker and vagrant base images (#4905)
PS D:\git\qmk> vagrant up
Bringing machine 'qmk_firmware' up with 'virtualbox' provider...
==> qmk_firmware: Importing base box 'generic/debian9'...
==> qmk_firmware: Matching MAC address for NAT networking...
==> qmk_firmware: Checking if box 'generic/debian9' is up to date...
==> qmk_firmware: Setting the name of the VM: qmk_qmk_firmware_1557597214168_36242
==> qmk_firmware: Fixed port collision for 22 => 2222. Now on port 2200.
==> qmk_firmware: Clearing any previously set network interfaces...
==> qmk_firmware: Preparing network interfaces based on configuration...
    qmk_firmware: Adapter 1: nat
==> qmk_firmware: Forwarding ports...
    qmk_firmware: 22 (guest) => 2200 (host) (adapter 1)
==> qmk_firmware: Running 'pre-boot' VM customizations...
==> qmk_firmware: Booting VM...
==> qmk_firmware: Waiting for machine to boot. This may take a few minutes...
    qmk_firmware: SSH address: 127.0.0.1:2200
    qmk_firmware: SSH username: vagrant
    qmk_firmware: SSH auth method: private key
    qmk_firmware:
    qmk_firmware: Vagrant insecure key detected. Vagrant will automatically replace
    qmk_firmware: this with a newly generated keypair for better security.
    qmk_firmware:
    qmk_firmware: Inserting generated public key within guest...
    qmk_firmware: Removing insecure key from the guest if it's present...
    qmk_firmware: Key inserted! Disconnecting and reconnecting using new SSH key...
==> qmk_firmware: Machine booted and ready!
==> qmk_firmware: Checking for guest additions in VM...
==> qmk_firmware: Running provisioner: shell...
    qmk_firmware: Running: inline script
    qmk_firmware: /bin/sh: 1: /vagrant/util/qmk_install.sh: not found
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
````

Upon further inspection, I can see that the mount point `/vagrant` does not even exist:

```
PS D:\git\qmk> vagrant ssh
vagrant@debian9:~$ ls /
bin   dev  home        initrd.img.old  lib32  libx32      media  opt   root  sbin  sys  usr  vmlinuz
boot  etc  initrd.img  lib             lib64  lost+found  mnt    proc  run   srv   tmp  var  vmlinuz.old
```

zvecr, 2019-05-11:
I just rebuilt my vagrant box and was not able to reproduce this issue, however i will continue to dig.

@igorrafael Do you have a `~\.vagrant.d\Vagrantfile` file at all?

drashna, 2019-05-14:
This should be fixed now.  Closing. 

If it's not, please re-open.

closed 2019-05-14 by drashna