qmk/qmk_firmware #5868: Planck rev6 unresponsive after flash

rvolosatovs, 2019-05-14:
## Describe the Bug

After flashing my planck rev6 it became unresponsive(worked fine before).
There is no more sound when the keyboard is plugged in and it is not recognised as a keyboard in MacOS or Linux.
I am able to reflash the keyboard by pressing the `reset` button at the back of the keyboard, but that does not help.
On Linux I see in `journalctl`:
```
<will add later>
```

## System Information

 - Keyboard:  
   - Revision (if applicable):  rev6
 - Operating system:
 - AVR GCC version: avr-gcc (GCC) 7.4.0

 - ARM GCC version: arm-none-eabi-gcc --version

 - QMK Firmware version: 0.6.373

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

drashna, 2019-05-14:
Sounds like you may have flashed the wrong firmware on it. 

Try reflashing the BIN file.

rvolosatovs, 2019-05-15:
Reflashing is the first thing I tried and I did so already multiple times. Here's the log:
```
make clean && make planck/rev6:default:dfu-util
QMK Firmware 0.6.373
Deleting .build/ ... done.
QMK Firmware 0.6.373
Making planck/rev6 with keymap default and target dfu-util

arm-none-eabi-gcc (GCC) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck/rev6/matrix.c                                                           [OK]
Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/planck/planck.c                                                                [OK]
Compiling: keyboards/planck/rev6/rev6.c                                                             [OK]
Compiling: keyboards/planck/keymaps/default/keymap.c                                                [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/process_keycode/process_audio.c                                                  [OK]
Compiling: quantum/process_keycode/process_clicky.c                                                 [OK]
Compiling: quantum/audio/audio_arm.c                                                                [OK]
Compiling: quantum/audio/voices.c                                                                   [OK]
Compiling: quantum/audio/luts.c                                                                     [OK]
Compiling: quantum/process_keycode/process_music.c                                                  [OK]
Compiling: quantum/encoder.c                                                                        [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/chibios/suspend.c                                                        [OK]
Compiling: tmk_core/common/chibios/timer.c                                                          [OK]
Compiling: tmk_core/common/chibios/bootloader.c                                                     [OK]
Compiling: tmk_core/common/chibios/printf.c                                                         [OK]
Compiling: tmk_core/common/chibios/eeprom_stm32.c                                                   [OK]
Compiling: tmk_core/common/chibios/flash_stm32.c                                                    [OK]
Compiling: tmk_core/common/bootmagic.c                                                              [OK]
Compiling: tmk_core/common/mousekey.c                                                               [OK]
Compiling: tmk_core/common/command.c                                                                [OK]
Compiling: tmk_core/protocol/chibios/usb_main.c                                                     [OK]
Compiling: tmk_core/protocol/chibios/main.c                                                         [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: tmk_core/protocol/chibios/usb_driver.c                                                   [OK]
Compiling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt1.c                                [OK]
Compiling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.c                             [OK]
Compiling: lib/chibios/os/rt/src/chsys.c                                                            [OK]
Compiling: lib/chibios/os/rt/src/chdebug.c                                                          [OK]
Compiling: lib/chibios/os/rt/src/chtrace.c                                                          [OK]
[... cut ...]
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 52608
Download        [=========================] 100%        52608 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
make planck/rev6:default:dfu-util  12.77s user 3.82s system 62% cpu 26.507 total
```

rvolosatovs, 2019-05-18:
I was able to fix this by downgrading GCC version.
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

willbush, 2019-06-14:
This is probably not you're issue. However, I wanted to share I get the exact same error here due to a faulty mono-directional usb-c port on my rev6 planck. The following is the log from `journalctl` where I unplug the usb cable from my keyboard (on the usb-c side), and plug it in the other way, and then replug it in the other way around.

    Jun 14 00:26:36 nixos kernel: usb 7-1: USB disconnect, device number 2
    Jun 14 00:26:36 nixos acpid[984]: input device has been disconnected, fd 26
    Jun 14 00:26:36 nixos acpid[984]: input device has been disconnected, fd 27
    Jun 14 00:26:36 nixos acpid[984]: input device has been disconnected, fd 28
    Jun 14 00:26:36 nixos acpid[984]: input device has been disconnected, fd 29
    Jun 14 00:26:49 nixos kernel: usb 7-1: new full-speed USB device number 3 using xhci_hcd
    Jun 14 00:26:49 nixos kernel: usb 7-1: device descriptor read/64, error -71
    Jun 14 00:26:49 nixos kernel: usb 7-1: device descriptor read/64, error -71
    Jun 14 00:26:50 nixos kernel: usb 7-1: new full-speed USB device number 4 using xhci_hcd
    Jun 14 00:26:50 nixos kernel: usb 7-1: device descriptor read/64, error -71
    Jun 14 00:26:50 nixos kernel: usb 7-1: device descriptor read/64, error -71
    Jun 14 00:26:50 nixos kernel: usb usb7-port1: attempt power cycle
    Jun 14 00:26:51 nixos kernel: usb 7-1: new full-speed USB device number 5 using xhci_hcd
    Jun 14 00:26:51 nixos kernel: usb 7-1: Device not responding to setup address.
    Jun 14 00:26:51 nixos kernel: usb 7-1: Device not responding to setup address.
    Jun 14 00:26:51 nixos kernel: usb 7-1: device not accepting address 5, error -71
    Jun 14 00:26:51 nixos kernel: usb 7-1: new full-speed USB device number 6 using xhci_hcd
    Jun 14 00:26:51 nixos kernel: usb 7-1: Device not responding to setup address.
    Jun 14 00:26:51 nixos kernel: usb 7-1: Device not responding to setup address.
    Jun 14 00:26:52 nixos kernel: usb 7-1: device not accepting address 6, error -71
    Jun 14 00:26:52 nixos kernel: usb usb7-port1: unable to enumerate USB device
    Jun 14 00:27:11 nixos kernel: usb 7-1: new full-speed USB device number 8 using xhci_hcd
    Jun 14 00:27:11 nixos kernel: usb 7-1: New USB device found, idVendor=feed, idProduct=6060, bcdDevice= 0.06
    Jun 14 00:27:11 nixos kernel: usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    Jun 14 00:27:11 nixos kernel: usb 7-1: Product: Planck
    Jun 14 00:27:11 nixos kernel: usb 7-1: Manufacturer: OLKB
    Jun 14 00:27:11 nixos kernel: usb 7-1: SerialNumber: 0
    Jun 14 00:27:11 nixos kernel: input: OLKB Planck as /devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb7/7-1/7-1:1.0/0003:FEED:6060.0009/input/input26
    Jun 14 00:27:11 nixos kernel: hid-generic 0003:FEED:6060.0009: input,hidraw5: USB HID v1.11 Keyboard [OLKB Planck] on usb-0000:09:00.3-1/input0
    Jun 14 00:27:11 nixos kernel: input: OLKB Planck Mouse as /devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb7/7-1/7-1:1.1/0003:FEED:6060.000A/input/input27
    Jun 14 00:27:11 nixos kernel: input: OLKB Planck System Control as /devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb7/7-1/7-1:1.1/0003:FEED:6060.000A/input/input28
    Jun 14 00:27:11 nixos kernel: input: OLKB Planck Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb7/7-1/7-1:1.1/0003:FEED:6060.000A/input/input29
    Jun 14 00:27:11 nixos kernel: input: OLKB Planck Keyboard as /devices/pci0000:00/0000:00:07.1/0000:09:00.3/usb7/7-1/7-1:1.1/0003:FEED:6060.000A/input/input30
    Jun 14 00:27:11 nixos kernel: hid-generic 0003:FEED:6060.000A: input,hidraw6: USB HID v1.11 Mouse [OLKB Planck] on usb-0000:09:00.3-1/input1
    Jun 14 00:27:11 nixos kernel: hid-generic 0003:FEED:6060.000B: hiddev99,hidraw7: USB HID v1.11 Device [OLKB Planck] on usb-0000:09:00.3-1/input2
    Jun 14 00:27:11 nixos systemd-logind[1010]: Watching system buttons on /dev/input/event18 (OLKB Planck)
    Jun 14 00:27:11 nixos systemd-logind[1010]: Watching system buttons on /dev/input/event20 (OLKB Planck System Control)
    Jun 14 00:27:11 nixos systemd-logind[1010]: Watching system buttons on /dev/input/event22 (OLKB Planck Keyboard)
    Jun 14 00:27:11 nixos systemd-logind[1010]: Watching system buttons on /dev/input/event21 (OLKB Planck Consumer Control)

I haven't yet tried flashing my planck in nixos and I came here hoping to get some clue on how to do that because when I try (in nix shell) I get `arm-none-eabi-gcc` command not found.

closed 2019-06-17 by drashna

uqs, 2021-08-11:
This isn't really fixed, is it? Downgrading the compiler will get harder and harder as time passes by. I'm currently stuck with GCC 10 or 11, and they produce broken firmware for the Preonic.

https://config.qmk.fm/#/preonic/rev3/LAYOUT_ortho_5x12 is on something like GCC 6.3.1 and doesn't suffer from this (yet).

uqs, 2021-08-11:
For anyone else coming here with a broken preonic or planck, the problem seems to be the code for the RGB stuff. Disabling both RGBLIGHT_ENABLE and RGB_MATRIX_ENABLE makes GCC 11 produce code that properly starts up the keyboard.

nickfyson, 2022-07-28:
PSA - for anyone who lands here after facing this problem after using the ZSA configurator, make sure the standard/glow switch is set appropriately!

[image]

Presumably this determines the same thing raised in the comment above, about whether the RGB specific codes are included, and if you pick glow with a non-RGB board things get very weird!