qmk/qmk_firmware #5869: document issue of faq_debug.md

ymzcdg, 2019-05-14:
1.Is "unless you throw in extra fruit salad of options" a joke? I can't understand this sentence.(This maybe a naive question)
2."setting in Power **Management property** tab" position of the stars may wrong.
3."oops, wrong order!" is telling us the right order is "test the size, erase the block", right?Seems the auther is a humorous person.
!!!!.Thanks for answer.

noroadsleft, 2019-05-14:
1. The fruit salad line means you have to send a lot of extra data to instruct a DFU tool to write into a bootloader. I'm not actually sure that's relevant here.
2. You are correct. The formatting should be `**Power Management** property` instead of `Power **Management property**`.
3. Not sure I understand what you're saying here.

Also it seems like the list nesting around this part of the page is wrong.

ymzcdg, 2019-05-15:
1.The "fruit salad" maybe useful. Without "salad", my understanding is that whatever you send, DFU tool always can't write bootloader.
3.Will **make DFU** test the size before erasing the block?
!!!.appreciate you

> 1. The fruit salad line means you have to send a lot of extra data to instruct a DFU tool to write into a bootloader. I'm not actually sure that's relevant here.
> 2. You are correct. The formatting should be `**Power Management** property` instead of `Power **Management property**`. Reading it over, I don't think the word `property` needs to be here.
> 3. Not sure I understand what you're saying here.
> 
> Also it seems like the list nesting around this part of the page is wrong.

drashna, 2019-05-15:
For `make`, yes, it will check the size of the firmware before flashing it. It will error out and not attempt the flash, if it's too large.

That said, DFU is actually sane, and won't flash a firmware that's too large, even if you try to force it.  It will error out.   Same with Teensy Loader.  

The only bootloader that needs the sanity check is Caterina (Pro Micros), since it will allow you to flash a firmware that is too large, and wipe out the bootloader.  I know from PERSONAL experience.

ymzcdg, 2019-05-16:
custom_quantum_functions.md
1. “which are called when the system is board is idled and when it wakes up, respectively.” maybe too much "is".
2. Can I understand "Tapping Term" as "name of a key"?
3. What does the “Underglow” mean? I can' find it in webster dictionary.
4. "dual function keys would be greatly improved by different timeouts than `LT` keys, or because some keys may be easier to hold than others." I can't understand this sentence. Three words:"timeouts" "LT" "hold" baffle me. Maybe these words have some meaning I don't know.
!!!. Thanks

fauxpark, 2019-05-16:
`TAPPING_TERM` refers to the window of time a keypress gets in order to be considered a "tap"; after that it is a "hold". It's not very well named... see #4637.

drashna, 2019-05-22:
> custom_quantum_functions.md
> 
> 1. “which are called when the system is board is idled and when it wakes up, respectively.” maybe too much "is".
lol, whoops, yup.  that's my fault, I'm sure.   Should be:
"which are called when the system board is idled and when it wakes up, respectively

> 2. "use the non noeeprom versions" can't understand.

For the `rgblight` functions, there is the `_noeeprom` version of the function, which does not write to the EEPROM (persistent storage), and the "regular" versions, which to write to EEPROM.  

EEPROM writes are limited (it's high, but ..... should still be used sparingly).   However, since the function is writing to eeprom, you would want to save the settings. 

> 3. Can I understand "Tapping Term" as "name of a key"?

It's the setting for deteriming between a tap and a hold.  It's the number of milliseconds for that.  Less than is a tap, more than is a hold. 

Mostly, this is use for Mod-Tap, Layer-Tap, and Tap Dance.  

> 4. What does the “Underglow” mean? I can' find it in webster dictionary.

It's another name for RGB Light.   Specifically, it creates a "glow" under the board, since the LEDs are usually on the bottom side of the keyboard, facing down.  

Also, another name that could be use is "mood lighting". 

You can kind of see why it's called that because of: 
![Planck](https://raw.githubusercontent.com/qmk/qmk_firmware/3774a7fcdab5544fc787f4c200be05fcd417e31f/keyboards/planck/keymaps/yang/planck-with-rgb-underglow.jpg}

> 5. "dual function keys would be greatly improved by different timeouts than `LT` keys, or because some keys may be easier to hold than others." I can't understand this sentence. Three words:"timeouts" "LT" "hold" baffle me. Maybe these words have some meaning I don't know.

This is for the "custom tapping term" or "per key tapping term", right? 

This is .... complicated.   

Basically, this feature allows you to specify a different "tapping term" per key.  So `LT(1, KC_A)` can have a different tapping term than `LT(2, KC_SPC)` or `MT(MOD_LCTL, KC_Z)`

Basically, it's easier to hold "space" with your thumb, than "A" or "Z" with your pinky.  So, a lower tapping term for those keys are very useful, but a longer tapping term for SPC is just fine. 

"LT" is `Layer Tap`: 
```
LT(layer, kc) - momentarily activates layer when held, and sends kc when tapped. Only supports layers 0-15.
```

The "timeout" is the "Tapping Term".  It's the timeout for determining if the press was a hold or a tap.  

And yeah, this is one of the more technical and difficult sections.  (you should see the PR where I tried to get this working.  It was a nightmare.  and fauxpark as an amazing sounding board, and help a LOT in getting it working). 

>    !!!. Thanks

No, thank you for putting in the work to get this translated!! it's amazing.  And the work is immensely appreciated!!

noroadsleft, 2019-05-22:
Fixed image link:

> You can kind of see why it's called that because of:
> [image]

closed 2019-06-04 by mechmerlin