qmk/qmk_firmware #5879: build error when disabling debug print, in ergodox_ez layouts

whitehorsesoft, 2019-05-15:
# General Description
Disabling debug printing results in a build error, for all keymaps for the _ergodox_ez_ keyboard.
# Detailed Description / Repro Steps
The QMK docs detail config options, among which are [features that can be disabled in `config.h`](https://docs.qmk.fm/#/config_options?id=features-that-can-be-disabled). The top feature in this section is the debugging feature, which is disabled via `#define NO_DEBUG`.

The ability to disable this feature is corroborated by this same line existing as a comment in `config.h` on line 127.

However, when this line is uncommented (that is to say, `#define NO_DEBUG` is defined to disable the debugging feature), and the `make` command is used in accordance with [instructions in the QMK docs](https://docs.qmk.fm/#/getting_started_make_guide?id=more-detailed-make-instructions), the build fails and an error is thrown.

For example, this is the result when attempting to build the included "vim" layout, although the same result occurs with any _ergodox_ez_ layout, including custom ones:

> ▒▒@▒▒▒▒▒ ~/qmk_firmware> make ergodox_ez:vim
> QMK Firmware 0.6.373
> WARNING:
>  Some git sub-modules are out of date or modified, please consider running:
>  make git-submodule
>  You can ignore this warning if you are not compiling any ChibiOS keyboards,
>  or if you have modified the ChibiOS libraries yourself. 
> 
> Making ergodox_ez with keymap vim
> 
> avr-gcc (GCC) 9.1.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> Compiling: keyboards/ergodox_ez/matrix.c                                                           In file included from <command-line>:
> ./keyboards/ergodox_ez/config.h:127: error: "NO_DEBUG" redefined [-Werror]
>   127 | #define NO_DEBUG
>       | 
> <command-line>: note: this is the location of the previous definition
> cc1: all warnings being treated as errors
>  [ERRORS]
>  | 
>  | 
>  | 
> make[1]: *** [tmk_core/rules.mk:372: .build/obj_ergodox_ez_vim/matrix.o] Error 1
> Make finished with errors
> make: *** [Makefile:553: ergodox_ez:vim] Error 1

Needless to say, this does not occur if `#define NO_DEBUG` is left commented out in `config.h`:

> ▒▒@▒▒▒▒▒ ~/qmk_firmware> make ergodox_ez:vim
> QMK Firmware 0.6.373
> WARNING:
>  Some git sub-modules are out of date or modified, please consider running:
>  make git-submodule
>  You can ignore this warning if you are not compiling any ChibiOS keyboards,
>  or if you have modified the ChibiOS libraries yourself. 
> 
> Making ergodox_ez with keymap vim
> 
> avr-gcc (GCC) 9.1.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> Compiling: keyboards/ergodox_ez/matrix.c                                                            [OK]
> Compiling: drivers/avr/i2c_master.c                                                                 [OK]
> Compiling: keyboards/ergodox_ez/ergodox_ez.c                                                        [OK]
> Compiling: keyboards/ergodox_ez/keymaps/vim/keymap.c                                                [OK]
> Compiling: quantum/quantum.c                                                                        [OK]
> Compiling: quantum/keymap_common.c                                                                  [OK]
> Compiling: quantum/keycode_config.c                                                                 [OK]
> Compiling: quantum/debounce/eager_pr.c                                                              [OK]
> Compiling: quantum/process_keycode/process_unicode.c                                                [OK]
> Compiling: quantum/process_keycode/process_unicode_common.c                                         [OK]
> Compiling: quantum/color.c                                                                          [OK]
> Compiling: quantum/rgblight.c                                                                       [OK]
> Compiling: drivers/avr/ws2812.c                                                                     [OK]
[... 31 lines cut ...]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
> Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
> Linking: .build/ergodox_ez_vim.elf                                                                  [OK]
> Creating load file for flashing: .build/ergodox_ez_vim.hex                                          [OK]
> Copying ergodox_ez_vim.hex to qmk_firmware folder                                                   [OK]
> Checking file size of ergodox_ez_vim.hex                                                            [OK]
>  * The firmware size is fine - 28992/32256 (3264 bytes free)

# System Information
 - Keyboard:  ErgoDox_EZ
   - Revision (if applicable): V3 (Batch 3a)
 - Operating system: Arch Linux 5.0.13
 - AVR GCC version: 9.1.0

 - ARM GCC version: 9.1.0

 - QMK Firmware version: 0.6.373

 - Any keyboard related software installed? 
   - None

fauxpark, 2019-05-15:
`NO_DEBUG` is always defined unless `CONSOLE_ENABLE = yes`:

https://github.com/qmk/qmk_firmware/blob/7209266ea1bd8b6bce4d468ea016689602499239/tmk_core/common.mk#L121-L126

So if you're trying to free up some flash space, you're out of luck here - it's already freed.

whitehorsesoft, 2019-05-15:
@fauxpark good catch! That closes this issue, but it seems to me that it runs contrary to the documentation (even [the rules.mk documentation](https://docs.qmk.fm/#/getting_started_make_guide?id=rulesmk-options) specifically says to define `NO_DEBUG`).

Have you found anything in the documentation to indicate otherwise? Basically I'm wondering if I should pursue having the documentation be updated.

fauxpark, 2019-05-15:
I think the docs are correct - note that the `NO_DEBUG` and `NO_PRINT` stuff is under the `CONSOLE_ENABLE` "heading". There is an implication here that they are only relevant if Console is enabled, but it's not exactly clear, I'll grant you that.

whitehorsesoft, 2019-05-15:
Thank you for the additional comments. I'm marking this as closed.

closed 2019-05-15 by whitehorsesoft