qmk/qmk_firmware #5891: Space Cadet LALT triggers alt modifier in some applications

jon-grangien-gaia, 2019-05-16:
## Describe the Bug
When using `KC_LAPO` with `LAPO_KEYS KC_LALT, KC_LSFT, KC_9` (left alt = alt on hold, open parens on tap), applications that look for the alt key seem to trigger an incompatibility with the mod-tap functionality on left alt.

Example: in the chrome address bar editing the url text, tapping works fine and produces an open parens, but in this text input area alt takes precedence and the key rarely produces an open parens on tap.

Another easy example is VS Code. Testing on windows, first press triggers alt that marks the items in the top menu (File, Edit etc), and only the second press works by producing the tap configured key.

Should be easy to try to reproduce, if not I will update the firmware to latest master and give more information. 

Config defines I have, not sure if related:
TAPPING_TERM 135
PERMISSIVE_HOLD
TAPPING_FORCE_HOLD

## System Information
 - Keyboard:  
   - Revision (if applicable):  Let's Split rev2
 - Operating system: Windows
 - Software compiling system: Linux
 - AVR GCC version: gcc8 I think

 - QMK Firmware version: Master branch from a few days ago (sorry I'm at work, will update if not reproducable)

jon-grangien-gaia, 2019-05-16:
@XScorpion2

XScorpion2, 2019-05-16:
It's not really a bug, this is just how Alt & Space Cadet work in general. Space Cadet always registers the modifier on press (alt in this case) so it is responsive for use with other keys. Alt triggers menus in applications, so it's not really useful for everyday typing with this setup. I'd suggest switching to TapDance in this case as it delays the initial key press until it can determine which method to use.

jon-grangien-gaia, 2019-05-17:
Ah I see, I hadn't got that. That makes it less useful as you say, but still good to have the options.

closed 2019-05-17 by jon-grangien-gaia