qmk/qmk_firmware #5902: [Request] Add at90usb support for Split Keyboards

drashna, 2019-05-17:
Specifically, the Split Keyboard code is hard coded to the ATmega32u4, at least for serial communication. 

It would be nice to expand that out to include other MCUs.  Especially the at90usb1286. 

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

Zubon102, 2019-05-21:
I second this! The Teensy 2.0++ is a very common and useful MCU and it would be great to be able to use it for serial communication.

drashna, 2020-10-25:
This is being addressed in an open PR and I can confirm it is working. Closing.

closed 2020-10-25 by drashna