qmk/qmk_firmware #5910: Using Bluetooth with the atmega328p as the only output method (no v-usb)

mhmd98, 2019-05-18:
This is a a question but could possibly be a bug report as i am new to qmk and not sure if i am missing something or if what i am trying to do is not fully supported (why not both :p)
What i am trying to do is to create a macro pad (6x6) using an arduino mini (3.3v, 8Mhz) with the RN42 bluetooth module.
I got a few errors here and there, beginning with:
**"USART configuration is needed."** When i set **BLUETOOTH = RN42**
from what i understand that is due to the missing USART configuration (obviously) for the atmega328p in config_common.h

[https://github.com/qmk/qmk_firmware/blob/d85110b6ec5fcdf3ef88b41909ce739b017abbea/quantum/config_common.h#L274-L296](url)

So i just changed it to:
```
/* USART configuration */
#ifdef BLUETOOTH_ENABLE
#   ifdef __AVR_ATmega328P__
#      define SERIAL_UART_BAUD 115200
#      define SERIAL_UART_DATA UDR1
#      define SERIAL_UART_UBRR (F_CPU / (8UL * SERIAL_UART_BAUD) - 1)
#      define SERIAL_UART_RXD_VECT USART1_RX_vect
#      define SERIAL_UART_TXD_READY (UCSR1A & _BV(UDRE1))
#      define SERIAL_UART_INIT() do { \
            /* baud rate */ \
            UBRR1L = SERIAL_UART_UBRR; \
            /* baud rate */ \
            UBRR1H = SERIAL_UART_UBRR >> 8; \
            /* enable TX */ \
            UCSR1B = _BV(TXEN1); \
            /* 8-bit data */ \
            UCSR1C = _BV(UCSZ11) | _BV(UCSZ10); \
            sei(); \
        } while(0)
#   else
#       error "USART configuration is needed."
#   endif
#endif
```
I am not sure if configuration is correct and it's probably not but i just wanted to move on and check if there were more errors waiting to appear and indeed there were:

```
Compiling: tmk_core/common/avr/suspend.c                                                           In file included from lib/lufa/LUFA/Drivers/USB/USB.h:383:0,
                 from tmk_core/protocol/lufa/lufa.h:49,
                 from tmk_core/common/avr/suspend.c:16:
lib/lufa/LUFA/Drivers/USB/Core/USBMode.h:260:7: error: #error USB_DEVICE_ONLY is not available for the currently selected microcontroller model.
      #error USB_DEVICE_ONLY is not available for the currently selected microcontroller model.
       ^
lib/lufa/LUFA/Drivers/USB/Core/USBMode.h:271:6: error: #error The currently selected device, USB mode or architecture is not supported.
     #error The currently selected device, USB mode or architecture is not supported.
      ^
In file included from lib/lufa/LUFA/Drivers/USB/Core/AVR8/../USBInterrupt.h:60:0,
                 from lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:56,
                 from lib/lufa/LUFA/Drivers/USB/Core/USBController.h:150,
                 from lib/lufa/LUFA/Drivers/USB/Core/USBTask.h:47,
                 from lib/lufa/LUFA/Drivers/USB/USB.h:386,
                 from tmk_core/protocol/lufa/lufa.h:49,
                 from tmk_core/common/avr/suspend.c:16:
lib/lufa/LUFA/Drivers/USB/Core/AVR8/../AVR8/USBInterrupt_AVR8.h:84:4: error: expected identifier before ‘}’ token
    };
    ^
In file included from lib/lufa/LUFA/Drivers/USB/Core/USBController.h:150:0,
                 from lib/lufa/LUFA/Drivers/USB/Core/USBTask.h:47,
                 from lib/lufa/LUFA/Drivers/USB/USB.h:386,
                 from tmk_core/protocol/lufa/lufa.h:49,
                 from tmk_core/common/avr/suspend.c:16:
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:114:5: error: #error No PLL prescale value available for chosen F_USB value and AVR model.
    #error No PLL prescale value available for chosen F_USB value and AVR model.
     ^
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h: In function ‘USB_Detach’:
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:190:5: error: ‘UDCON’ undeclared (first use in this function)
     UDCON  |=  (1 << DETACH);
     ^
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:190:5: note: each undeclared identifier is reported only once for each function it appears in
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:190:22: error: ‘DETACH’ undeclared (first use in this function)
[... 76 lines cut ...]
lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.h:407:22: error: ‘USBE’ undeclared (first use in this function)
     USBCON &= ~(1 << USBE);
                      ^
tmk_core/common/avr/suspend.c: In function ‘power_down’:
tmk_core/common/avr/suspend.c:102:7: error: ‘USB_DeviceState’ undeclared (first use in this function)
   if (USB_DeviceState == DEVICE_STATE_Configured) return;
       ^
tmk_core/common/avr/suspend.c:102:26: error: ‘DEVICE_STATE_Configured’ undeclared (first use in this function)
   if (USB_DeviceState == DEVICE_STATE_Configured) return;
                          ^
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:369: recipe for target '.build/obj_HOT36BT_default/common/avr/suspend.o' failed
make[1]: *** [.build/obj_HOT36BT_default/common/avr/suspend.o] Error 1
Make finished with errors
```
so i just added PROTOCOL = VUSB to my rules.mk and added usbconfig.h and i was able to compile.
the thing is:
i am using and arduino pro mini to achive 6x6 layout so i am using pins 2 all the way to 13 which makes for total of 12 pins. I don't really understand much about how vusb works but doesn't it use some of the same pins i am using for cols and rows? also how does QMK know which ports to use for TX and TX when BLUETOOTH is equal to RN42 and is there a way to specify those pins? and last but not least is there a way to compile without using vusb or at least make sure that vusb won't interfering with the rows and cols if i keep it? also bare in mind that vusb wouldn't actually compile if i set F_CPU to 8MHz as this is not a legal value so i have to set it to 16MHz which is different from my actual clock speed so would  that be an issue?.
please also notice that i already have the pcb with pins 0 and 1 connected to the RN42 and 2-7 as rows and 8 to 13 as cols so i can't really change much when it comes to pins.

[config.txt](https://github.com/qmk/qmk_firmware/files/3194165/config.txt)
[rules.txt](https://github.com/qmk/qmk_firmware/files/3194166/rules.txt)

mhmd98, 2019-06-08:
@drashna was wondering if you could just inform me if what i am trying to do is possible or if i should just give it up. I've seen that you have answered similar question so i thought i would ask you directly. Thanks

drashna, 2019-06-08:
Unfortunately, I'm not really familiar with the bluetooth stuff. 

And I suspect that it will take a lot of digging to get this working.

mhmd98, 2019-06-08:
Yeah it's seems so, Thank you for your time.

closed 2019-06-08 by mhmd98