qmk/qmk_firmware #5922: RGB Under glow and matrix on two pins

al3ph, 2019-05-19:
Hi, I've designed a custom numpad, and I've wired the underglow RGB LEDS and backlight RGB LEDs to different pins (they are WS2812B's), but I can see anyway of making it work as there is only one RGB_DI_PIN defined, is there anyway of making this work ? 

Or are the two lighting systems mutually exclusive ?

cheers

fauxpark, 2019-05-19:
You probably want the [RGB Matrix](https://docs.qmk.fm/#/feature_rgb_matrix?id=ws2812-avr-only) feature for the inswitch LEDs, and RGB Light (as you're using now) for the underglow. I think RGB matrix has/will have the ability to manage underglow too.

al3ph, 2019-05-19:
Yup I'm using already using RGBMatrix, the problem is I need two pins, one for the leds for the keys and on for the leds underneath the board.

They are separate RGB led strings.

fauxpark, 2019-05-19:
Ah, yeah. The way that both of these features work (and I think the underlying drivers in fact), there can only be a single data pin driving the LEDs.

Paging @XScorpion2, any thoughts?

XScorpion2, 2019-05-19:
ya, pretty much all the led drivers we have all assume there is only single one of each type in existence. So you could have 1 is driver and 1 ws driver, but not 2 ws drivers.

al3ph, 2019-05-19:
Ah darn, that makes sense, should have checked.. bit late now as it's at the PCB fab. :( On the plus sie I can probably run a wire between the two strings too connect them, I assume you can flag some of the LED's in the matrix as underglow, but it's not clear what pattern they'll use.

XScorpion2, 2019-05-19:
most of the effects in rgb matrix are based on the physical position of the leds that you configure. So they will smoothly blend in with each other. There is some work in progress that will allow you to run different effects per type of led, so the backlights could use one effect, while the underglow could us another. That wip pr is here: https://github.com/qmk/qmk_firmware/pull/5851

al3ph, 2019-05-19:
Cool, that's just the ticket, thanks for the help!

closed 2019-05-19 by al3ph