qmk/qmk_firmware #5930: KBD66/AMJ66 hardware revisions

FSund, 2019-05-20:
Turns out there are different hardware revisions of the PCB for the "KBD66" keyboard that has been sold through both Massdrop and KBDfans' own site. Some PCB's are labelled "AMJ66", and it seems like they have a different pin configuration than other boards. They used TMK originally.

Here is what I think is the official firmware for the AMJ66 labelled PCB's: https://github.com/AMJKeyboard/AMJ66

I have two "KBD66" PCB's, so I'll have a look at the differences and try different firmwares when have a chance, but just wanted to report this now.

There should perhaps be a warning about this in the KBD66 repo, since if I flash the KBD66 firmware to my AMJ66 board almost no keys work.

mechmerlin, 2019-05-20:
I wonder if this was KBDFan's attempt to make a board called a kbd66, which took several different kinds of PCBs. In fact the KBD66 also fits a clueboard 66. 

Perhaps this is like a TX60 which comes with a Jigon PCB, but can fit pretty much any standard 60% pcb. Maybe the AMJ66 should be in its own directory.

FSund, 2019-05-21:
I tried just copying the pins, layout and keymap from [judy2k's fork](https://github.com/judy2k/qmk_firmware/tree/kbd66/keyboards/kbd66/keymaps/default), and that seems to work fairly well. The only thing that doesn't work at the moment is the upper right key. I've tried adjusting the layout in `amj66.h`, but have been unable to fix the issue.

I have made a fork with my additions here:
https://github.com/FSund/qmk_firmware/tree/amj66/keyboards/kbdfans/amj66

I am not certain how KBD66 and AMJ66 should be organized, but for now I've put it in the kbdfans folder.

Should I open a pull request already, even though the work isn't done, or should I leave it as is in my own fork for now?

FSund, 2019-08-02:
Fixed by https://github.com/qmk/qmk_firmware/pull/5945

closed 2019-08-02 by FSund