qmk/qmk_firmware #5934: RGB Set Color APIs - Breaking Change discussion

XScorpion2, 2019-05-20:
There are a slew of set rgb apis in rgb matrix and rgb light that take individual r, g, b values all typedef to uint8_t. I'd like to rip all those APIs out and replace them with two sets of new APIs that take either the HSV struct or RGB struct and have that carry all the way through to the final drivers. (Maybe add and HSL API too?)

Ultimately this will allow us to swap out the underlying implementations of these structs more easily in the future for keyboards using 16 (48) bit RGB drivers as well as reduce how frequently we construct and these structs only to pass in the raw values, only to reconstruct it again later on in some cases. In addition, this sets us up to allow easier conversion to const & of these inputs for more performance and firmware size savings if we start allowing cpp features in the future.

Looking at the compiler output there is a performance/size catch for this, namely this requires a few more instructions as we are constructing a struct on the stack to pass in. However, this is only at the calling site where the construction occurs.

XScorpion2, 2019-05-20:
Also I'm half tempted to convert rgb matrix to cpp

GitWellBack, 2019-05-20:
It's also a good first step in helping us get a unified API that we can take into bidirectional HID comms and whatever front-end we end up creating for RGB layouts/animations.
We need to be mindful that AVR will have size AND speed issues with anything over 8 bit.
Definitely agree with a larger datapath, having said that - will be useful for the future.
Not so worried about the CPP angle - can cause a lot of issues for people new to the codebase than the quasi-c we have now, from what I've seen - although again, given the constant push towards ARM, we may be wise to keep it in view.

XScorpion2, 2019-05-20:
Migrating to cpp for pass by const & or pointer is how we should handle the larger data paths. Here is the avr gcc 4.6.4 output between the two methods. Pass individual values is 5 cycles, pass by const & or pointer is 6 cycles, pass by value is 8 cycles. While pointer and ref are the same at the call site and execution site, const & is a bit easier for users to work with on the code side of things. Though it's minor.

```c
  // pass individual values
  mov w1, -1  // 1 cycle
  mov w2, -128  // 1 cycle
  mov w0, w1  // 1 cycle
  bl hsv_to_rgb2(unsigned char, unsigned char, unsigned char)  // 2 cycle

  // pass by value
  ldd r22,Y+4 // 2 cycle
  ldd r23,Y+5 // 2 cycle
  ldd r24,Y+6 // 2 cycle
  rcall hsv_to_rgb(HSV)  // 2 cycle

  // pass by const &
  mov r24,r28 // 1 cycle
  mov r25,r29 // 1 cycle
  adiw r24,4    // 2 cycle
  rcall hsv_to_rgb(HSV const&)  // 2 cycle

  // pass by pointer
  mov r24,r28 // 1 cycle
  mov r25,r29 // 1 cycle
  adiw r24,1    // 2 cycle
  rcall hsv_to_rgb(HSV*)  // 2 cycle
```

GitWellBack, 2019-05-20:
Not my place to say - but I think const & is the way to go - I can see a lot of nested brackets becoming a thing if we go pass by pointer!

spidey3, 2020-05-21:
I think this is a great idea - in no small part because it will make it easier to shift the implementation to provide better / more correct support for RGBW leds...