qmk/qmk_firmware #5936: QMK Configurator site is missing multiple dz60 layouts

KarBoy2314PL, 2019-05-20:
Hey there,

Multiple layouts for dz60 are still missing. Is using LAYOUT_ALL a temporary solution?

For example, this one is missing:
http://www.keyboard-layout-editor.com/#/gists/156be10eb42f21d5f37546824483a767

also #4668 ?

That's all, I want them added, thx

noroadsleft, 2019-05-20:
> For example, this one is missing:
> http://www.keyboard-layout-editor.com/#/gists/156be10eb42f21d5f37546824483a767

This layout is `LAYOUT_60_b_ansi` with a 2u Backspace and a 6.25u Spacebar.

This file should get you started. The extra `Space` and `Backspace` keys won't be an issue, because there won't be switches present to make the electrical connection.

[dz60_karboy2314pl.json.txt](https://github.com/qmk/qmk_firmware/files/3199242/dz60_karboy2314pl.json.txt)

 
> also 4668 ?

That issue lost steam because it became apparent that supporting every layout the DZ60 supports wasn't possible from a labor standpoint. DZ60 supports *several hundred* layout combinations, and we just don't have the human resources to support that. We've shifted direction a bit on this, and have begun moving in a direction of supporting the "superset" of a layout, and then users can use a subset of that layout if their own isn't directly supported.

KarBoy2314PL, 2019-05-20:
> The extra Space and Backspace keys won't be an issue, because there won't be switches present to make the electrical connection.

that's what I was mostly worried about, thx for help

yanfali, 2019-05-31:
@KarBoy2314PL does this resolve your issue?

KarBoy2314PL, 2019-06-14:
> @KarBoy2314PL does this resolve your issue?

partially

drashna, 2019-06-17:
Closing then.  

Basically, if your layout isn't supported, `LAYOUT_all` should work, with a bit of guesswork or trial and error.

closed 2019-06-17 by drashna