qmk/qmk_firmware #5954: Keymap: ACTION_LAYER_TAP_KEY and Linux Compose Key

gg587998, 2019-05-22:
when i use the compose key i have to wait a substantial time after I press and release the compose key, then i can type the keys to be compose. otherwise nothing will be outputed.

## Describe the Bug

I am using ACTION_LAYER_TAP_KEY(1, KC_CAPS). I use the compose in linux to create special characters like ö and ä with compose+a+" or compose+o+". Where the " is ACTION_LAYER_TAP_KEY(1,KC_QUOTE). 

Recently i updated my repository and reflashed my unaltered keymap. Now i ran into the following problem: When I type one of the combinations from above. I need to wait maybe a few ms until i can type a+" to produce ä. This slows down my typing.
Even when i reduce the TAPPING_TERM nothing changes. I played around with the other tapping related setting to no avail.

## System Information

 - Keyboard:  
   - Revision (if applicable):  ergodox_ez
 - Operating system: manjaro linux
 - AVR GCC version: avr-gcc (GCC) 9.1.0

 - ARM GCC version: 

 - QMK Firmware version: 0.6.385

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2019-06-17:
I'm confused, are you having issues with CAPS or Quote?   Or both?

gg587998, 2019-06-26:
Hi with both so to speak.
in software (not on my ergodox keyboard) i remapped CAPS->linux compose, ' "->" '
i now press the combo CAPS+o+' , which gives me compose+o+", an creates the character ö

drashna, 2019-06-26:
Try adding `#define TAP_HOLD_CAPS_DELAY 0` to your keymap's `config.h`, and see if that helps.

gg587998, 2019-07-02:
thanks for the answer,  i already tried that before, it does not change anything

CharString, 2022-01-10:
@gg587998 Related to this. Do I understand correctly that there's no actual keycode for Compose? I remap right alt to compose with xmodmap, but I have to redo that every time I disconnect the keyboard. It would be nice if I could put that in QMK.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc

gg587998, 2023-01-28:
Sorry for the late reply, I was very busy and did not use the keyboard much.
I discovered the problem is not software related, maybe a problem with the
keyboard itself or my computer.

Please close ticket