qmk/qmk_firmware #5958: Newest version of arm-none-eabi-gcc on Arch Linux (9.1) causes an error with trying to make keymaps

Zawaken, 2019-05-22:
The newest version of arm-none-eabi-gcc causes an error that makes it impossible to make a keymap.

## Describe the Bug
```QMK Firmware 0.6.385
Making massdrop/ctrl with keymap default

arm-none-eabi-gcc (Arch Repository) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: tmk_core/common/arm_atsam/bootloader.c                                                  tmk_core/common/a
tmk_core/common/arm_atsam/bootloader.c:35:9: error: array subscript -4 is outside array bounds of 'uint32_t[1]' {aka
   35 |         *MAGIC_ADDR = BOOTLOADER_MAGIC;                         //Set magic number into RAM
In file included from tmk_core/common/arm_atsam/bootloader.c:19:
tmk_core/protocol/arm_atsam/md_bootloader.h:12:17: note: while referencing '_eram'
   12 | extern uint32_t _eram;
      |                 ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:372: .build/obj_massdrop_ctrl_default/common/arm_atsam/bootloader.o] Error 1```

<!-- A clear and concise description of what the bug is. -->

## System Information

 - Keyboard:  Massdrop CTRL
   - Revision (if applicable):  
 - Operating system: Linux (Arch)
 - AVR GCC version: 9.1.0
<!-- Run `avr-gcc --version` to find this out. -->
 - ARM GCC version: 9.1.0
<!-- Run `arm-none-eabi-gcc --version` to find this out. -->
 - QMK Firmware version: 0.6.385
<!-- Run `git describe --abbrev=0 --tags` to find this out. -->
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

<!-- Add any other relevant information about the problem here. -->
I was able to fix this with:
`pacman -U https://archive.archlinux.org/packages/a/arm-none-eabi-gcc/arm-none-eabi-gcc-8.3.0-1-x86_64.pkg.tar.xz` (might have to do it two times with just adding `.sig` to the end of that link if it doesn't work right away)

drashna, 2019-05-22:
Yeah, we've been seeing that.

Looks like we may need to change the script.

Specifically, one or more of these lines:
https://github.com/qmk/qmk_firmware/blob/master/util/linux_install.sh#L59-L61

Since I don't have Arch to test with.... it may need somebody else to test and configure the script

Zawaken, 2019-05-23:
So it would look more like this then?
```
elif grep ID /etc/os-release | grep -q 'arch\|manjaro'; then
	sudo pacman -U https://archive.archlinux.org/packages/a/avr-gcc/avr-gcc-8.3.0-1-x86_64.pkg.tar.xz
        pacman -U https://archive.archlinux.org/packages/a/arm-none-eabi-gcc/arm-none-eabi-gcc-8.3.0-1-x86_64.pkg.tar.xz
	sudo pacman -S \
		arm-none-eabi-binutils \
                arm-none-eabi-gcc \
		arm-none-eabi-newlib \
		avr-binutils \
		avr-libc \
		avr-gcc \
		base-devel \
		dfu-util \
		diffutils \
		gcc \
		git \
		python \
		unzip \
		wget \
                zip
```
I also assume that installing the 8.3 version of the other libraries connected to arm-none-eabi-gcc might be smart, although it wasn't needed to make it work for me.

Carlovan, 2019-05-26:
Hi,
I think using an outdated version of the compiler it's not an actual solution: it doesn't allow us to use new features in the future and is forces you to install that specific version (I use it for other stuff other than compiling QMK).

I should have find a solution to this problem, but I cannot try it because I haven't a Massdrop CTRL.

Just replace line **14** in file `tmk_core/protocol/arm_atsam/md_bootloader.h` from
```#define MAGIC_ADDR (uint32_t *)(&_eram - 4)```
to
```#define MAGIC_ADDR (uint32_t *)((intptr_t)(&_eram) - 4)```
The cast "hides" to the compiler the fact that `&_eram` is a pointer, so it should stop complaining about you accessing outside the "array".

@Zawaken I ask you to try it and report back.
Thank you

drashna, 2019-05-27:
The problem is that the older versions are NOT outdated. 

Ubuntu ships with 5.4.0, and the package manager does not offer you a newer version. 
Same goes with a number of the other compilers. 

Calling it outdated is ....problematic. 

And upgrading the code just so it will work on the latest compiler may not be the best option either, if it breaks things for the older versions.  

In this case, YES, using the older version is the solution. Otherwise, you're telling everyone that they have to install, track and update their gcc compilers manually.  

Additionally, the 9.1.0 has a number of other issues that I've seen reported, as well.  

So just blinding upgrading because it's the newest is also not an actual solution, either. 

Basically, there is no good option here.

drashna, 2019-05-27:
And to clarify, I'm not saying we shouldn't fix the code here...

What I'm saying is that any fixes need to work on both old versions of the compiler, and on the new versions.

Carlovan, 2019-05-27:
Right, thank you for your good explanation. I forgot that Arch always rushes with the updates...

Anyway my fix should work on older versions too, I'm just casting a pointer to `intptr_t` before performing the subtraction, and then recasting it to a pointer

Ronnie76er, 2019-05-31:
Just to let you know, I'm on arch and have a Massdrop Ctrl keyboard, and was facing the same issue. I made the code change suggested here to `md_bootloader.h`, and it compiled fine. I loaded it on to my keyboard, and everything is working as far as I can tell. Let me know if you want me to provide any more information.

doughsay, 2019-06-01:
I'm a new Massdrop CTRL owner and just ran into this as well. The above mentioned fix to `tmk_core/protocol/arm_atsam/md_bootloader.h` fixes the compilation issues. Should I open a PR with the above change?  Will it be accepted?

drashna, 2019-06-17:
this should be fixed now.  Closing.

closed 2019-06-17 by drashna