qmk/qmk_firmware #5977: Documentation clarifications

Wraul, 2019-05-25:
As part of implementing support for a new keyboard I had to do some digging into the documentation. This resulted in a few questions that I think could be clarified in the documentation.

1. Should the [layout macros](https://docs.qmk.fm/#/hardware_keyboard_guidelines?id=ltkeyboard_namehgt) be named `LAYOUT_ansi` or ex. `LAYOUT_tkl_ansi` to work with the layouts feature.
2. The [hardware options](https://docs.qmk.fm/#/config_options?id=hardware-options) specifies that setting `BACKLIGHT_PIN` to either pin `B5`, `B6` or `B7` will use hardware PWM. But the [backlight caveats](https://docs.qmk.fm/#/feature_backlight?id=caveats) also mentions pin `C6`. 
3. The [contributing documentation](https://docs.qmk.fm/#/contributing?id=documentation) states that one should standardise on `my_layers` and `my_keycodes`. But the actual template used when creating a new keyboard uses no layers enum and `custom_keycodes` for the keycodes.

fauxpark, 2019-05-25:
The layout macros need to match the layouts under `layouts/default` in both name and arguments (so a TLK ANSI layout should be named `LAYOUT_tkl_ansi` and take 87 arguments laid out in the standard TKL arrangement).

For the backlight pin, yeah, the config options page is most likely out of date in more than just that area. Hardware PWM is definitely supported on C6 on the ATmega32U4, and also D4 for the ATmega32A.

It doesn't *really* matter what these enums are named, since you'll rarely reference them directly, only their elements. But I think I'd prefer to standardise on `layer_names` and `custom_keycodes.`

drashna, 2019-06-17:
Was this still an issue? 

Eg, do we need to take action here, or was this okay to close this issue?

fauxpark, 2019-06-17:
Addressed point 2 in #6102 just now.

drashna, 2019-11-21:
I think these have all been dealt with.

closed 2019-11-21 by drashna