qmk/qmk_firmware #5982: OLED doesn't turn off after timeout

mattdibi, 2019-05-26:
OLED screen doesn't turn off after timout using the new OLED driver API (oled_driver.h)

Most probably I'm doing something wrong but I cannot understand what.

I'm building a new macropad design which uses an OLED module using SSD1306 over I2C. The code is [here](https://github.com/mattdibi/qmk_firmware/tree/yampad/keyboards/yampad). Everything is working fine except for the OLED screen turn off feature. From what I gather from the API code the OLED screen should turn off after 60s, but this doesn't happen. Am I using the API wrong?

Thanks for the help.

XScorpion2, 2019-05-26:
Got a link to the code?

XScorpion2, 2019-05-26:
Hehe, found the bug, on non-split keyboards, the matrix_scan code is returning that it changed every frame, thus why it doesn't turn off.

mattdibi, 2019-05-29:
I see it has been merged. I already tested the changes on my fork and it works flawlessly. Thank you very much! :D

Lenbok, 2019-07-06:
Can you reopen this? There are several boards for which the matrix_scan is not returning the change status, e.g. CRKBD (where I noticed it), Helix, Lily58... etc where the OLED timeout is not working.

drashna, 2019-07-07:
A possible interim solution is: https://github.com/qmk/qmk_firmware/pull/6128/commits/1521335c5bf8c850817e129342989663a3113051

And actually, this works better on split common, IMO.

XScorpion2, 2019-07-07:
Honestly I'm not sure this really is a bug that custom keyboards that forgo the us of split common also have to implement custom oled timeout work. The docs even call out the oled driver uses split common and it is recommended for keyboards to switch to it.

That being said, I could do a pass on the custom split impls to add oled timeout easy enough if you ask nicely =p

Lenbok, 2019-07-07:
I agree that the best way forward is for those split keyboards to migrate to split-common, and wouldn't want you to spend time on that when there is the workaround like @drashna posted. It may be worth adding a note to the OLED docs alongside the current split-common mention to note this as another reason to encourage boards to migrate. 

It's probably worth keeping this issue open until it is resolved one way or another (even if that is just to add a note to the docs), so that it's less likely that others open new issues about it.

XScorpion2, 2019-07-07:
hmm, note in the doc makes too much sense! lol will get on that.

XScorpion2, 2019-07-07:
https://github.com/qmk/qmk_firmware/pull/6276

drashna, 2019-08-22:
Is it okay if we close this?

Lenbok, 2019-08-23:
Yes, I think so.

closed 2019-08-24 by mattdibi