qmk/qmk_firmware #6002: Custom matrix_scan for some boards seem incorrect

LucidityCrash, 2019-05-28:
Specifically in redox_w/matrix.c but also in others the block in matrix_scan the uart_data array seems like it is not the right size, shouldn't it be MATRIX_COLS + 1, the for loop should use this too and with the comparison uart_data[10] == 0xE0) shouldn't it be MATRIX_COLS ?

drashna, 2019-06-17:
Yeah, I think it may need some cleanup.  

But somebody with a board would need to test the changes, I think.   
@mattdibi ?

mattdibi, 2019-06-17:
Thanks for the input!

I'll test the changes this weekend (if I'll remember to bring my Redox home from work...). Note that this may be a common problem in Mitosis-derived keyboards (like the Redox).

I'll update you in this thread.

LucidityCrash, 2019-06-17:
I've had a bit of a thought about this, your matrix is sent from the nrf5 as rows not columns isn't it ?  Given this basically works, and if you really were only looping over the first 10 columns things would be a tad broken I'd have thought ... I think the thing that confused me was the comment saying 14 columns.
Does that make sense ?

mattdibi, 2019-06-23:
Sorry for the delay.

Data are indeed sent by row and there's 11 rows, 10 actual the 11st is for signalling the end of transmission. No need to change anything in the code apart from the comments.

drashna, 2019-08-22:
Comments would be a good addition, though. :)

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc