qmk/qmk_firmware #6006: A better implementation of Grave Escape under macOS?

Morid1n, 2019-05-28:
Under macOS, Grave Escape (KC_GESC) currently makes accessing the backtick character impossible. I might have a possible solution.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

KC_GESC under macOS currently works as follows:

- Pressing ESC generates ESC.
- Pressing Shift + ESC generates the tilde (~) character.
- Pressing LGUI + ESC launches the Cmd + Backtick shortcut, which is like an Alt+Tab for windows contained in one app (a very useful feature).
- Option/Alt + ESC currently does nothing.

Since the Option/Alt + ESC shortcut is "free to use", I would like it to generate the backtick character (which I can't live without). This would basically solve the whole problem.

fauxpark, 2019-05-28:
So, unfortunately Control+<code>\`</code> does not produce <code>\`</code>, and since the way Grave Escape works is by simply sending a `KC_GRV` or `KC_ESC` without tweaking the held modifiers, it's a no-go. I got macOS to type a backtick with Alt+<code>`</code>, but this appears to be a dead key in certain circumstances so may not be 100% what you need.

```c
// placed underneath the GRAVE_ESC_SHIFT_OVERRIDE block in quantum.c
#ifdef GRAVE_ESC_ALT_FORCE
      // if ALT is pressed, KC_GRV is always sent
      if (get_mods() & MOD_MASK_ALT) {
        shifted = 1;
      }
#endif
```

Making this feature work properly will likely require a little bit of a rewrite to keep track of the held mods and disable/enable them when it comes time to send the Grave or Escape.

Morid1n, 2019-05-28:
Funny thing. If I hold Shift + Alt + KC_GESC, I get a Ŕ character, but if I add Control _after_ pressing those three, I get a backtick! Proof: ````````````. Spams the character though. 🙂

How much work would it be to do an `#ifdef` for Control?

Morid1n, 2019-05-28:
Another idea: since LShift + ESC are used to generate `~`, then LShift + LControl + ESC could be used for backtick ```.

drashna, 2019-06-17:
Do none of these help? 

https://docs.qmk.fm/#/feature_grave_esc?id=configuration

Also, there is: https://docs.qmk.fm/#/feature_grave_esc?id=caveats

fauxpark, 2019-06-17:
@drashna - the issue is that the modifiers controlling Grave/Esc aren't removed before sending it, which presents problems when there is a key combination mapped to it. Thus as I mentioned above, we need to unset the mods first and put them back afterwards. (I can see this being an issue for Alt and GUI on Windows)

drashna, 2019-06-17:
Ah, okay

Morid1n, 2019-10-30:
Any news on this? Tell me what I can do to help, if needed.

This function would free up 1 key on my 65s and make life so much easier.

fauxpark, 2019-10-30:
Honestly, I'm not convinced this is fixable, at least not without some serious side effects that would basically render it useless. The problem is that when `KC_GESC` is pressed with Alt or GUI, we first have to unset those modifiers in order to send the proper key combination. This will complete a "tap" event from the perspective of the OS, and on Windows (maybe Linux too) will change the focus, to the menu bar or the start menu respectively.

Morid1n, 2019-10-30:
What if another modifier was used for the backtick?

fauxpark, 2019-10-30:
Same deal, you have to release that Alt before sending either `KC_GRV` or `S(KC_GRV)`.

Morid1n, 2019-10-30:
Oh boy. I’m currently duplicating a key. GESC top left and a regular backtick+tilde top right, just for that backtick... such a waste.

noroadsleft, 2019-10-30:
I used to split time between macOS and Windows, and at the time, I had my top left key (60%) as a dedicated Escape key, and I modded my Left Ctrl key so it was `KC_LCTL` when held, but `KC_GRV` when tapped.

yanfali, 2019-10-30:
> I used to split time between macOS and Windows, and at the time, I had my top left key (60%) as a dedicated Escape key, and I modded my Left Ctrl key so it was `KC_LCTL` when held, but `KC_GRV` when tapped.

I do the inverse. LCTL_T(KC_ESC) on capslock and KC_GRV on the grave

Morid1n, 2019-10-30:
Seems like I’ll have to do something similar. Siding towards Caps Lock (which is CTRL for me) tap for ESC.

masterzen, 2019-11-09:
My issue specifically is the dead key backtick which is very helpful for international users. On a classic macos keyboard you get it with ALT+\`, but it isn't possible to get it with `KC_GESC`.  I still haven't found a good solution to combine all usage in a key.
In retrospect, I think what I'd like to have is (based on my usage):
* normal \` on single press
* dead key \` on ALT+\`
* normal ~ on SHIFT+\`
* ESC on CTRL/GUI+\`

I might as well use a `LCTL_T(KC_ESC)` on caps-lock (which is CTRL on my keyboards)...

20four80five, 2020-12-21:
>     * normal ` on single press
>
>     * dead key ` on ALT+`
> 
>     * normal ~ on SHIFT+`
> 
>     * ESC on CTRL/GUI+`

@masterzen this is exactly what I'm trying to do as well: 

- ESC key = back-tick `
- GUI + ESC key produces `Escape`
- SHIFT + ESC key produces `~`

masterzen, 2020-12-22:
@20four80five I made an implementation that works as you wish:

https://github.com/masterzen/qmk_firmware/blob/07159c2c1c651d7ae47b54182503c73b7b9d173e/keyboards/exclusive/e65/keymaps/masterzen/keymap.c#L113-L125

20four80five, 2020-12-23:
> @20four80five I made an implementation that works as you wish:
> 
> https://github.com/masterzen/qmk_firmware/blob/07159c2c1c651d7ae47b54182503c73b7b9d173e/keyboards/exclusive/e65/keymaps/masterzen/keymap.c#L113-L125

@masterzen thank you. I tried this but couldn't get it to work correctly.
The online QMK keyboard test indicated that the ESC key was pressed but not _unpressed_?

I also referred to the [Macros - QMK](https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/feature_macros) documentation but found that confusing when looking/comparing to you [keymap.c](https://github.com/masterzen/qmk_firmware/blob/07159c2c1c651d7ae47b54182503c73b7b9d173e/keyboards/exclusive/e65/keymaps/masterzen/keymap.c#L113-L125)

This is my current `keymap.c`:

```c
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT_66_ansi(KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, KC_INS, KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_DEL, KC_ENT, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_LCTL, KC_LALT, KC_LGUI, KC_SPC, KC_RGUI, KC_RCTL, MO(1), KC_LEFT, KC_DOWN, KC_RGHT),
	[1] = LAYOUT_66_ansi(KC_ESC, KC_BRID, KC_BRIU, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_MUTE, KC_VOLD, KC_VOLU, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_CAPS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS)
};
```

Any tips?

20four80five, 2020-12-25:
@masterzen - Happy holidays!

I finally managed to get this working! Let me know your thoughts.

Aside from what I mentioned earlier above:
>- ESC key = back-tick `
>- LeftGUI + ESC key produces `Escape`
>- SHIFT + ESC key produces `~` 

...I also wanted to be able to `Command + Shift + Escape` to cycle through windows of the same application.

🤞🏼 that this helps other folks looking to do the same thing on macOS

---

I'm still new to QMK - this is what I added to my `keymap.c`:
- indicated by: `// ADDED THIS`
- I didn't comment the `BACK_TICK` for the actual key in layout [0].
- I had to add `!(leftGui && leftShift)` because without it, I wasn't able to `Command + Shift + Escape` to cycle through windows of the same app.
- I also had to adjust `#define DEBOUNCE 5` to `#define DEBOUNCE 10` in my `config.h` as there was chatter.

```C
#include QMK_KEYBOARD_H

// ADDED THIS
enum my_keycodes {
	BACK_TICK = SAFE_RANGE,
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT_66_ansi(
		BACK_TICK, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, KC_INS, KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_DEL, KC_ENT, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_LCTL, KC_LALT, KC_LGUI, KC_SPC, KC_RGUI, KC_RCTL, MO(1), KC_LEFT, KC_DOWN, KC_RGHT),
	[1] = LAYOUT_66_ansi(
		KC_TRNS, KC_BRID, KC_BRIU, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_MUTE, KC_VOLD, KC_VOLU, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_CAPS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS)
};

// ADDED THIS
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case BACK_TICK: {
      static uint8_t kc;
			static bool leftGui;
			static bool leftShift;

      if (record->event.pressed) {
        leftGui = get_mods() & (MOD_BIT(KC_LGUI));
				leftShift = get_mods() & (MOD_BIT(KC_LSFT));

        if ((leftGui) & !(leftGui && leftShift)) {
          del_mods(MOD_BIT(KC_LGUI));
          kc = KC_ESC;
        } else {
          kc = KC_GRV;
        }

				register_code(kc);

				if ((leftGui) & !(leftGui && leftShift)) {
					register_code(KC_LGUI);
				}

      } else {
        unregister_code(kc);
      }
    }
  }
  return true;
}
```

Soundtoxin, 2021-02-13:
I noticed that right shift + esc still makes a ~ with this key configuration (KC_GESC), contrasting with how the same feature works on the Pok3r's factory firmware. It would be nice if only left shift did it. My work-around in the meantime was to:
1) put ` on another layer or two where esc is so I can hit it when needed (same as Pok3r)
2) on a different layer put a shift + escape combo press where the escape key is, because it's otherwise impossible to use shift-escape, which is needed to leave the "passthrough" mode in qutebrowser.

On my Pok3r I used to just hit right shift and escape for those situations.
I was gonna open an issue about this, but this one seems close enough. I'm not on macOS, fyi.

alanhoyle, 2021-07-01:
I am noticing this as well as system-wide keystroke is getting suppressed.  

On macOS/OSX going back at least as long as I've used it, option+command+Esc (⌥⌘ Esc) should bring up the Force Quit... dialog box, however, it's sending option+command+` (⌥⌘`) instead.

kbrock, 2022-01-28:
I'm pretty sure you want to define `GRAVE_ESC_ALT_OVERRIDE`.
So command back tick and command alt escape both work.

But I'm using VIAL right now and don't know how to get that option into my firmware (may need to compile myself)

alanhoyle, 2022-02-01:
@kbrock, I'd been unaware of VIAL before.  Thanks for pointing it out.  Perhaps since it's open source (unlike VIA), there would be a way to get it implemented there?  I recently bought a VIA-compatible keyboard and it's much easier than recompiling/flashing with regular QMK, albeit not as flexible.

artt, 2022-04-29:
I'm new to QMK and don't really get 100% of the conversation here, but would this work:

* Set Left Cmd to MO(9)
* On layer 9, map each key to Cmd + Key, except for the escape key, which maps to Cmd + `.

closed 2022-08-29 by zvecr