qmk/qmk_firmware #6007: Mouse support for ARM/ChibiOS

eyemyth, 2019-05-28:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

drashna, 2019-06-17:
Do you mean like MOUSEKEYS?  

If so, I can confirm that they do work on ChibiOS/ARM.  

Or do you mean the Pointing device stuff? 
If so, that does appear to work on ARM.  

So, I'm not sure what the issue here is?

eyemyth, 2019-06-17:
I mean pointing device. I'm trying to get a TrackPoint working.

drashna, 2019-08-22:
Ah, okay.

drashna, 2020-02-14:
Specifically, to clarify, pointing devices do work with ARM.  It's what mousekeys leverages. 

https://docs.qmk.fm/#/feature_pointing_device

The issue is, that you will have to handle the communication, as well as the pointing device reports.    You can see an example of this in my 40percent/nano keymap.  

The other question here, is how is the trackpoint being connected?  I2C, analog, SPI, etc.

closed 2020-06-13 by stale[bot]