qmk/qmk_firmware #6014: Using encoder push button directly wired to controller?

quarterto, 2019-05-30:
So, I have a Levinson, with an encoder added by me, A and B directly wired to pins F4 and F5. It's working great. I also wired C to B5, in the hope that I could use the encoder push button as an extra key. How can I read the state of that pin and trigger a key using it? Is this a use case for something like `DIRECT_PINS`? Doesn't seem exactly the same.

quarterto, 2019-05-30:
i have this sort-of-working with:

```c
void matrix_init_user(void) { setPinInputHigh(B5); }

void matrix_scan_user(void) {
  if (!readPin(B5)) {
    tap_code(KC_MUTE);
  }
}
```

this doesn't debounce at all, and i'm not sure how to do that since it's not a matrix key. is there a way to treat it as a matrix key? can i use `debounce` and `raw_matrix` from outside of `matrix.c`?

quarterto, 2019-05-30:
okay, i've got this working for my needs using a flag for the state of the encoder. it would be nice to have it treated as a matrix key, but this will do for now:

```c
void matrix_init_user(void) { setPinInputHigh(B5); }

static bool encoder_pressed;

void matrix_scan_user(void) {
  if (readPin(B5)) {
    encoder_pressed = false;
  } else {
    if (!encoder_pressed) {
      tap_code(KC_MUTE);
    }
    encoder_pressed = true;
  }
}
```

closed 2019-05-30 by quarterto

quarterto, 2019-05-30:
actually, would this be a good addition to `encoder.c`? if there's a `ENCODER_BUTTON_PAD` defined, add this code, and have it call `encoder_pressed_kb`/`encoder_pressed_user`?

thmjpr, 2019-09-23:
It would be good in encoder.c, but another method is now available for anyone looking:
- You can define the pins as a dip switch: #define DIP_SWITCH_PINS { B2, B5 }
- Now these are available for press/release callbacks in keymap.c, most people would only need to use press for the action, for example:

`
void dip_switch_update_user(uint8_t index, bool active) 
{ 
 switch (index) {
        case 0:		//ENC0_SW
            if(active) //ENC0 pressed
				{ 
				tap_code(KC_MUTE);
				} 
			else //ENC0 released
{
				//audio_off(); 
				}
            break;
`

dlawle, 2022-01-12:
Just have a question, but this topic may be dead. 

I have two encoders on a custom board, as outlined as the following: 

```
#define ENCODERS_PAD_A { D7, B5 }
#define ENCODERS_PAD_B { D4, B4 }
#define ENCODER_RESOLUTION 4
```

[I have them set up as outlined in the QMK documentation here.](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_encoders.md#hardware)

Is there any chance that I can get the push button functioning without C having it's own dedicated pin? If not, could I test by wiring a jumper from the C pin on the encoder to a free pin?

thmjpr, 2022-01-12:
Yes you can use C (push switch) in your switch matrix, as you would any
other normal two pin switch, and it won't require a dedicated pin.
This is because the encoder switch is usually completely isolated from the
encoder circuitry itself.

If you instead wire C to a free pin, then the other side of the C switch
would go to ground.

dlawle, 2022-01-12:
Is that true of encoders with dedicated sw pins?

[Looks like I’m using this one here. ](https://aws1.discourse-cdn.com/arduino/original/4X/d/4/6/d4660aa66768f23a3dece96d1ba2be53300ff1e4.png) 

I’ve attached a small jumper from the marked sw pin to an unused pin and implemented in keymap.c:

```
void matrix_init_user(void) { setPinInputHigh(B5); }

void matrix_scan_user(void) {
  if (!readPin(B5)) {
    tap_code(KC_MUTE);
  }
}
```

And still have no action on press. 

Is there another define that needs to be made elsewhere?

Apologies for all the questions. I’m a bit new to this and learning a mile a minute!

thmjpr, 2022-01-12:
The photo you show is isolated yes, Switch and Gnd labeled pins on the left
side would not be connected to anything else until you connect them.

If you are using a dedicated pin for a switch then use the dip_switch
function as I've explained above.

dlawle, 2022-01-12:
Apologies for turning this closed issue into a troubleshooting session. 

[Photo of the back of my PCB](https://user-images.githubusercontent.com/39495124/149240701-21abe545-ee6d-4ec8-8869-6a4c4fae68a6.jpg)

I have tried connecting the red pin to E6 (empty pin) and blue pin to GND (and vice versa) with jumpers, and still am not having any luck. 

In my config.h:

```
/* Encoders */

#define ENCODERS 1

#define ENCODERS_PAD_A { D7, B5 }
#define ENCODERS_PAD_B { D4, B4 }
#define ENCODER_RESOLUTION 4

#define DIP_SWITCH_PINS { E6 }
```
and in my keymap.c:

```
void dip_switch_update_user(uint8_t index, bool active) 
{ 
 switch (index) {
        case 0:		//ENC0_SW
        if(active) //ENC0 pressed
		{ 
			tap_code(KC_A);
		} 
	else //ENC0 released
                {
			//do nothing
		}
            break;
	}
}
```

(I changed the tap code to KC_A in order to have easier to validate testing)

I've set up a small return code in debugging, but I still don't see any return on press. Any ideas on what I'm missing? Or is it more possible that I just have a bad encoder?

thmjpr, 2022-01-14:
Looks right to me. The chance of having a bad encoder would be very low.
I think the issue is there is no "E6" that is likely "B6".

You can also check with a multimeter, measure voltage of B6/E6 when
operating, it should be 3.3V or 5V. Then when you press on the encoder it
should drop to 0V.