qmk/qmk_firmware #6047: Unable to change rgb_matrix settings before a "wait_ms" or a "reset_keyboard"

cabouffard, 2019-06-01:
Hey,

First of all, I am not sure if this is a bug or if it's a misunderstanding on my end about how this work. I'm kind of new to QMK so please be gentle ;)

I'm a recent owner of the DZ60RGB and I've been trying to flash my keyboard so that before resetting it, I change the color of my board such as I'm about to know I've pressed the right key combination. 

I was inspired by [this bit](https://github.com/qmk/qmk_firmware/blob/master/keyboards/dztech/dz60rgb/keymaps/matthewrobo/keymap.c#L243-L248). Unfortunately, I couldn't get this to work. The keyboard never changes color before `reset_keyboard`. It seems to eat the command and reset it before the keyboard as the time to change its color.

I've tried the following combinations:

```c
if (timer_elapsed32(key_timer) >= 500) {
  rgb_matrix_sethsv(128, 255, 255);
  wait_ms(150);
  rgb_matrix_sethsv(40, 255, 255);
  reset_keyboard();
}
```

And the keyboard pauses and resets without ever changing color. I've [contacted the author](https://github.com/qmk/qmk_firmware/pull/5993#issuecomment-497579080) and he's been having the same issue. He's also been having the same issue on the `CTRL` board.

Lastly, I tried to execute `SEND_STRING('test')` in between so that I'm truly aware that it works for every step:

```c
if (timer_elapsed32(key_timer) >= 500) {
  SEND_STRING('changing to cyan')
  rgb_matrix_sethsv(128, 255, 255);
  wait_ms(150);
  SEND_STRING('changing to yellow')
  rgb_matrix_sethsv(40, 255, 255);
  reset_keyboard();
}
```

and it's properly sending the string to the active window.

Is this a bug or am I having a misunderstanding on how this should work?

Thanks!

XScorpion2, 2019-06-01:
RGB Matrix needs to flush the new hsv value to the LED drivers and cannot do so during wait_ms

drashna, 2019-06-11:
Would calling `keyboard_task` or `matrix_scan` fix that ?

aka, where is the flush called?

XScorpion2, 2019-07-19:
Since the rgb matrix code was updated to do a single render frame calculation over several frames, you would probably need several calls to task or scan to trigger the flush. 
An option would be to extern define the flush function and call it right after you set the values each time:
```extern void rgb_matrix_update_pwm_buffers(void);

void in_some_function(void)
{
  if (timer_elapsed32(key_timer) >= 500) {
    SEND_STRING('changing to cyan')
    rgb_matrix_sethsv(128, 255, 255);
    rgb_matrix_update_pwm_buffers();
    wait_ms(150);
    SEND_STRING('changing to yellow')
    rgb_matrix_sethsv(40, 255, 255);
    rgb_matrix_update_pwm_buffers();
    reset_keyboard();
  }
}
```

drashna, 2019-08-22:
Awesome.  Can confirm that this works. :)

XScorpion2, 2019-08-22:
@drashna I couldn't remember if rgb_matrix_update_pwm_buffers was defined in the header or not is all.

drashna, 2019-09-06:
It is.  I've implemented this, and it works perfectly.

cabouffard, 2019-11-22:
Hey sorry for being so late to the party. I had totally forgotten about this, that's my bad!

Unfortunately, I'll be the party pooper as it doesn't work for me. I've tried what you have suggested and currently have the following

(Please do note that I'm aware that I'm going overkill on `                 rgb_matrix_update_pwm_buffers();`. I just wanted to make sure that it would be called at the right time.

```c

...

extern void rgb_matrix_update_pwm_buffers(void);

...

bool process_record_user(uint16_t keycode, keyrecord_t* record)
{
    static uint32_t key_timer;

    switch (keycode) {
        case REBOOT:
            if (record->event.pressed) {
                key_timer = timer_read32();
            } else {
                if (timer_elapsed32(key_timer) >= 500) {
                    rgb_matrix_mode(RGB_MATRIX_BREATHING);
                    rgb_matrix_update_pwm_buffers();
                    rgb_matrix_sethsv(88, 255, 100);
                    rgb_matrix_update_pwm_buffers();
                    wait_ms(2000);
                    rgb_matrix_update_pwm_buffers();
                    rgb_matrix_sethsv(0, 255, 100); // To see if this would work after the wait_ms but it doesn't.
                    rgb_matrix_update_pwm_buffers();
                    reset_keyboard();
                } else {
                    rgb_matrix_sethsv(0, 255, 100);
                }
            }

            return false;
        default:
            return true;
}
```

By doing the above, the keyboard waits 2 seconds before doing the reset, but the RGB colors and the animations aren't changing still.

At this point, this just becomes a fun little thing, it's not a requirement that it works. I'd just like for my keyboard to turn green and start breathing as it loads the firmware in, instead of apparently just "freezing" as it does it.

fauxpark, 2019-11-23:
The breathing animation will not work while the keyboard is in bootloader mode, as QMK is not running at that point (this is why the RGB always freezes when you hit `RESET`, the animation task is just running continuously). Some of the ISSI chips have the ability to do breathing all on their own, but the drivers don't make use of it. It may be possible to send the autobreath command manually, though.

drashna, 2019-12-01:
Yeah. There are two issues here, actually. 

I think on the `wait_ms` function, it will cause the animation to pause/stop processing until it's done, anyways.  And then once you reset to bootloader, it's no longer running the task. 

The first issue could be fixed on ARM by using a thread for RGB Matrix tasks.  
The second issue, @fauxpark is correct about.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc

0e4ef622, 2023-05-10:
I recently found this thread trying to accomplish the same thing, and while `rgb_matrix_update_pwm_buffers` didn't work for me, using [deferred execution](https://docs.qmk.fm/#/custom_quantum_functions?id=deferred-execution) did. Here's what my code looks like:

```c
uint32_t reset_keyboard_callback(uint32_t trigger_time, void *cb_arg) {
    reset_keyboard();
    return 0;
}

// somewhere in process_record_user
rgb_matrix_mode_noeeprom(RGB_MATRIX_SOLID_COLOR);
rgb_matrix_sethsv_noeeprom(0, 255, 255);
// call reset_keyboard() 50 ms later
defer_exec(50, reset_keyboard_callback, NULL);
```