qmk/qmk_firmware #6051: Not sure how to do unicode

YamiFrankc, 2019-06-02:
Hello, I am in Windows 10, and have Wincompose installed.

I am trying to add some unicode characters to my keyboards by adding `UC(0x00F1)` to a key, however when I type the key it only gives me "00f1".
I am not really sure if I need to enable something elsewere, do I need to do something else?

Thanks

drashna, 2019-06-02:
Have you set the unicode mode? 

https://docs.qmk.fm/#/feature_unicode?id=switching-input-modes

The simplest way would be to add `#define UNICODE_SELECTED_MODES UC_WINC` to your config.h, and compile. 

Also, make sure that you repo is up to date, and that winCompose is up to date (there was a change not long ago to WinCompose and a PR to reflect that.

YamiFrankc, 2019-06-02:
Thanks, that #define was it.

closed 2019-06-02 by YamiFrankc

drashna, 2019-06-02:
Awesome, glad to hear it!