qmk/qmk_firmware #6053: How to make a key that sends a keycode on tap, but enters layer and holds mod on hold?

bouncepaw, 2019-06-02:
Hi. I am making my first QMK keyboard and I'm confused.

Imagine a key that sends `a` when simply tapped, but presses `LCTL` and enters layer 0 on hold (it returns to the previous layer after that).

I have found some similar things in the documentation:

`LT(0, KC_A)` would send `a` and enter layer 0, but wouldn't hold `LCTL`. 

`MT(MOD_LCTL, KC_A)` would send `a` and hold `LCTL`, but wouldn't enter layer 0.

Both of them do not fit me. Could you help me please?

drashna, 2019-06-02:
Well, `LM` is close to what you want.  

But you'd want have to write a custom macro for this. 

Something like: 
```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    static uint16_t fnx_layer_timer;

    switch (keycode){
        case KC_FNX:                                    
            if(record->event.pressed){
                fnx_layer_timer = timer_read();
                layer_on(_FL);
                register_mod(MOD_LCTL);
            } else {
                layer_off(_FL);
                unregister_mod(MOD_LCTL);
                if (timer_elapsed(fnx_layer_timer) < get_tapping_term(keycode)) {  
                    tap_code(KC_A);
                }
            }
        return false;
    }
    return true;
}

bouncepaw, 2019-06-02:
Thank you! That is exactly what I wanted. I also have one question: do I have to create a separate timer for each such key or not?

drashna, 2019-06-02:
Just add more of the `uint16_t` variables.  Make sure they're static, otherwise, the value won't be preserved.

bouncepaw, 2019-06-03:
I see. Thank you for help.

I'm closing the issue.

closed 2019-06-03 by bouncepaw