qmk/qmk_firmware #6061: RGBLIGHT_LIMIT_VAL not respected by many functions e.g. rgblight_setrgb(r, g, b)

rickard-von-essen, 2019-06-03:
In the [documentation](https://beta.docs.qmk.fm/features/feature_rgblight#configuration) `RGBLIGHT_LIMIT_VAL` is described with:

> The maximum brightness level

But reading the code this only limits the brightness when using [`sethsv`](https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L127-L131) not when using the [`setrgb`](https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L133-L137) (or any of the functions calling it).

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I suggest that this check is moved to the [`setrgb`](https://github.com/qmk/qmk_firmware/blob/master/quantum/rgblight.c#L133-L137) function (which is called by `sethsv`). This would make `RGBLIGHT_LIMIT_VAL` an overall limit of the RGB brightness which I think was the intention.

If you agree I can create a PR to address this.

drashna, 2019-06-17:
That's a good point, and yeah, should be addressed in some way.  

@XScorpion2 comments, ideas?

XScorpion2, 2019-06-17:
RGBLIGHT_LIMIT_VAL is HSV only, so it makes sense the setrgb functions to do not respect that define. I do agree not having a global "brightness limit" is definitely a good idea, though how to go about doing that correctly will be interesting as it has to play nice with hsv > rgb conversion, linear lighting table, and raw rgb values uniformly.

Thinking about it, I think you would probably want to do the following steps:

-  Remove / deprecate the RGBLIGHT_LIMIT_VAL define in favor of a clearer global brightness define
-  Remove the linear light table from hsv_to_rgb
-  Add scale8(X, BRIGHTNESS_LIMIT) to the lower level drivers
-  Add linear light table using the results from the previous brightness limit

That should keep the colors correct with how they are viewed today and keep things consistent across rgblight & rgbmatrix.

danielo515, 2019-07-03:
So the only way to control the rgb bright is by setting it to lower values?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc