qmk/qmk_firmware #6070: KC_KEYMAP undefined when running `make planck/rev6:default`

wavebeem, 2019-06-04:
It seems like `keyboards/planck/planck.h` only defines KC_KEYMAP (and thus KC_LAYOUT_ortho_4x12) for non-revision 6 plancks. I want to use KC_KEYMAP in my personal 
layout, but I have to copy the macro from the file instead.

I'm running `make planck/rev6:default` and it complains the macro is undefined.

I'd be happy to assist with a PR but I wasn't sure if the omission of KC_KEYMAP from the `#else` branch after `#ifdef __AVR__` was intentional or not.

Note that I have built and flashed several keymaps, so I'm confident my local setup is working well, it's just that the header file for planck is missing this macro in certain circumstances.

This is the modified `keymap.c` I'm trying to use:

```c
#include QMK_KEYBOARD_H

#define _BASE 0
#define _SYMB 1
#define _NAV 2

#define KC_RSET RESET
#define KC_SYMB MO(_SYMB)
#define KC_NAV MO(_NAV)
#define KC_____ KC_TRNS
#define KC_XXXX KC_NO

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

  [_BASE] = KC_KEYMAP(
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ESC  , Q    , W    , E    , R    , T    , Y    , U    , I    , O    , P    , BSPC ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    TAB  , A    , S    , D    , F    , G    , H    , J    , K    , L    , SCLN , QUOT ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    LSFT , Z    , X    , C    , V    , B    , N    , M    , COMM , DOT  , SLSH , ENT  ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    XXXX , XXXX , XXXX , LCTL , LALT , LGUI , SPC  , SYMB , NAV  , XXXX , XXXX , XXXX
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
  ),

  [_SYMB] = KC_KEYMAP(
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , 1    , 2    , 3    , 4    , 5    , 6    , 7    , 8    , 9    , 0    , ____ ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , EXLM , AT   , HASH , DLR  , PERC , CIRC , AMPR , ASTR , LPRN , RPRN , PIPE ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , LCBR , RCBR , PLUS , UNDS , TILD , GRV  , MINS , EQL  , LBRC , RBRC , BSLS ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , ____ , ____ , ____ , ____ , ____ , ____ , ____ , ____ , ____ , ____ , ____
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
  ),

  [_NAV] = KC_KEYMAP(
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , F1   , F2   , F3   , F4   , MUTE , MPLY , HOME , UP   , END  , ____ , ____ ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , F5   , F6   , F7   , F8   , VOLU , MNXT , LEFT , DOWN , RGHT , DEL  , ____ ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , F9   , F10  , F11  , F12  , VOLD , MPRV , PGDN , PGUP , TAB  , CAPS , ____ ,
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
    ____ , ____ , ____ , ____ , ____ , ____ , ____ , RSET , ____ , ____ , ____ , ____
// ,-----,------,------,------,------,------,------,------,------,------,------,------,
  )

};
```

fauxpark, 2019-06-04:
Unfortunately, layouts that automatically prefix keycodes with `KC_` such as this one are discouraged as they don't play well with the Configurator, and if you're creating a keymap manually you have to define custom keycodes beginning with `KC_` do get things like `MT()`s and `LT()`s etc. They should really be removed, but there are still people using them out in the wild.

My guess is that this particular layout was omitted from the rev6 for precisely this reason.

wavebeem, 2019-06-05:
Oh, I see. I guess I'll just add KC_ back to all my layouts and bump up the grid size. Thanks for the info.

closed 2019-06-05 by wavebeem