qmk/qmk_firmware #6072: dfu-programmer erase error: "atmel.c:866: Flash NOT blank beginning at 0x0."

rgreenblatt, 2019-06-05:
This is the same error in [this issue](https://github.com/qmk/qmk_firmware/issues/2549), but that was never resolved. The full error can be found [here](https://pastebin.com/raw/5VT805cF). I just got the keyboard (gergo), and I was able to flash it before. The issue started after I tried to flash the board on a different computer which also happened to have a different version of dfu-programmer (6.2 while I had previously flashed it with 7.2). I am guessing this isn't relevant. I have since tried both computers, different usb cable, and various versions of dfu-programmer, but all consistantly fail in the same way.

The other issue sugggests that the problem can be resolved by ISP flashing the board. I don't have another board to do this with, so before I do this it would be great to know if there are other things I should try. Thanks for the help.

marhalloweenvt, 2019-06-06:
In my experience, when i encounter that error, the only and best way is erase all the chip and burn fresh bootloader. Then, problem solved. I have tried many different ways, but still not work at all. I think you should buy a spare Pro Micro (obviously it's cheap) and make it become a handful ISP programmer. It will help you a lot in your keyboard career.

rgreenblatt, 2019-06-06:
Thanks for the help. If this is a common issue, maybe it is worthwhile to document? It isn't obvious to me where in the docs this should go, but I would be happy to contribute a PR otherwise.

rgreenblatt, 2019-06-09:
ISP flashing fixed the issue.

marhalloweenvt, 2019-06-10:
> ISP flashing fixed the issue.

It's good to hear this.

@drashna i think we should close this issue. Thank in advanced

closed 2019-06-11 by drashna