qmk/qmk_firmware #6094: DZTech DZ60rgb build fails on Windows.

sli, 2019-06-07:
Building the DZ60rgb firmware on Windows 10 fails with this error:

```bash
Compiling: lib/chibios/os/hal/lib/streams/nullstreams.c                                             [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v7m.S                           [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/usr/bin/sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [tmk_core/rules.mk:354: .build/obj_dztech_dz60rgb_default/obj.txt] Error 1
Make finished with errors
make: *** [makefile:553: dztech/dz60rgb:default] Error 1
```

I've tested most of the DZ60rgb keymaps, running `make clean` between each test build, and all of them failed at the same spot with the same error. Other firmwares -- specifically the Keebio Iris -- build successfully. I made sure to update my QMK utils before I tested everything.

Here are my version numbers:

* QMK: 0.6.397, pulled shortly before filing this issue
* avr-gcc: `(AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0`
* arm-none-eabi-gcc: `(GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]`

moyi4681, 2019-06-13:
Yes, I can confirm fixing the issue by installing the newer packages.
Just get the libnewlib-dev (2.4.0.20160527-4) and libnewlib-arm-none-eabi (2.4.0.20160527-4) *.deb files, then install them:

sli, 2019-06-13:
This is not an issue on Linux, I tested building all the DZTech keymaps on Linux shortly after filing this issue and they built fine.

drashna, 2019-06-13:
Are you using MSYS, or WSL? 

If you're using WSL, then yeah, manually installing the deb files is the only solution that I've seen, aside from manually updating things.

drashna, 2019-06-17:
@sli how were you compiling?   WSL, MSYS2?

tyler274, 2021-01-14:
Running into this on a different board with mingw64

sli, 2021-04-07:
Haven't messed with QMK builds directly since 2019 since I got Gitlab's CI/CD to do my builds for me, sorry, but I'm using MSYS2.

Kosaro, 2021-04-22:
Same problem trying to build dactyl manuform using QMK MSYS

Kosaro, 2021-04-22:
I fixed it by removing gmake from my path. I found where it was using `type -a gmake`. It happened to by installed as part of strawberry perl, so I uninstalled that and now it works.