qmk/qmk_firmware #6101: add support for IS31FL3741

miraclehoshi, 2019-06-08:
## Feature Request Type

- [ ] Core functionality
- [●] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

QMK uses the RGB matrix and supports IS31FL3731, 3733, 3736, and 3737.

But those four ICs support less than 100 keys of RGB.

The IS31FL3741 from the same company supports 117 RGB LEDs.

This means that you can add RGB to a single IC on a 104- to 108-key keyboard

Can you add support for IS31FL3741?

I usually use a full size keyboard.

Using multiple ICs complicates the design.

drashna, 2019-06-08:
Well, we have a number of the ISSI chips already supported.   It should be a simple matter of taking one of them, and porting the code over for the 3741.

miraclehoshi, 2019-06-08:
> Well, we have a number of the ISSI chips already supported. It should be a simple matter of taking one of them, and porting the code over for the 3741.

I think so too, but I am very poor in programming knowledge. Also, even if I make it, it's not official support.

yiancar, 2019-06-08:
IF someone creates a PCB with that chip we can talk about it:)

miraclehoshi, 2019-06-08:
> IF someone creates a PCB with that chip we can talk about it:)

I tried to design a PCB using IS31FL3741.
But I found out a few hours ago that QMK didn't support it.

So I wrote issue. If QMK officially support for IS31SL3741, I must use this.

yiancar, 2019-06-08:
You can give it a go and we can work with it together:)

Wilba6582, 2019-06-08:
I'll build a breakout PCB, confirm datasheet.

GitWellBack, 2019-06-08:
I'm happy to supply existing code for this - but it will be a few weeks before I'm allowed/in position.

yiancar, 2019-06-08:
Quite happily this driver doesnt even have a sync pin so even less complexity!

e11i0t23, 2019-08-03:
@GitWellBack is there any update on this?

4pplet, 2020-01-18:
Any progress on the IS31FL3741?

melgeek001365, 2020-06-20:
> we can

> IF someone creates a PCB with that chip we can talk about it:)

Hello. We have used the ISSI3741 for many board already. it simple and better than 3733 and 3731.

PR #9201 (2020-06-20): Support IS31FL3741 and IS31FL3741A.

## Description

## Types of Changes

- [x] Core
- [ ] Bugfix
- [x] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Closes #6101

## Checklist

- [x] My code follows the code style of this project.
- [x] My change requires a change to the documentation.
- [x] I have updated the documentation accordingly.
- [x] I have read the [**CONTRIBUTING** document](https://docs.qmk.fm/#/contributing).
- [x] I have added tests to  [... cut]
closed 2020-06-20 by tzarc