qmk/qmk_firmware #6106: Can't compile planck/rev6 keymaps that uses LAYOUT_planck_mit

joecabezas, 2019-06-09:
fresh git clone, tried to compile included keymap named `vifon` but fails

here is the output of `make planck/rev6:vifon`

```
➜  qmk_firmware git:(master) ✗ make planck/rev6:vifon
QMK Firmware 0.6.398
Making planck/rev6 with keymap vifon

arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck/rev6/matrix.c                                                           [OK]
Compiling: keyboards/planck/planck.c                                                                [OK]
Compiling: keyboards/planck/rev6/rev6.c                                                             [OK]
Compiling: keyboards/planck/keymaps/vifon/keymap.c                                                 keyboards/planck/keymaps/vifon/keymap.c:36:13: error: implicit declaration of function 'LAYOUT_planck_mit' [-Werror=implicit-function-declaration]
     [_QW] = LAYOUT_planck_mit( /* Qwerty */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:36:13: error: initializer element is not constant
keyboards/planck/keymaps/vifon/keymap.c:36:13: note: (near initialization for 'keymaps[0][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:42:13: error: initializer element is not constant
     [_CM] = LAYOUT_planck_mit( /* Colemak */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:42:13: note: (near initialization for 'keymaps[1][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:48:13: error: initializer element is not constant
     [_PP] = LAYOUT_planck_mit( /* Pure Pro */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:48:13: note: (near initialization for 'keymaps[2][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:54:14: error: initializer element is not constant
     [_PPG] = LAYOUT_planck_mit( /* Pure Pro: Gaming */
              ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:54:14: note: (near initialization for 'keymaps[3][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:60:13: error: initializer element is not constant
     [_NM] = LAYOUT_planck_mit( /* Numeric */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:60:13: note: (near initialization for 'keymaps[4][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:66:13: error: initializer element is not constant
     [_LW] = LAYOUT_planck_mit( /* LOWER */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:66:13: note: (near initialization for 'keymaps[5][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:72:13: error: initializer element is not constant
     [_RS] = LAYOUT_planck_mit( /* RAISE */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:72:13: note: (near initialization for 'keymaps[6][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:78:13: error: initializer element is not constant
     [_DL] = LAYOUT_planck_mit( /* DUAL */
             ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:78:13: note: (near initialization for 'keymaps[7][0][0]')
keyboards/planck/keymaps/vifon/keymap.c:84:14: error: initializer element is not constant
     [_DYN] = LAYOUT_planck_mit( /* special */
              ^~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/vifon/keymap.c:84:14: note: (near initialization for 'keymaps[8][0][0]')
keyboards/planck/keymaps/vifon/keymap.c: In function 'enable_gaming_layer':
keyboards/planck/keymaps/vifon/keymap.c:97:5: error: implicit declaration of function '_delay_ms' [-Werror=implicit-function-declaration]
     _delay_ms(100);
     ^~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:369: recipe for target '.build/obj_planck_rev6_vifon/keyboards/planck/keymaps/vifon/keymap.o' failed
make[1]: *** [.build/obj_planck_rev6_vifon/keyboards/planck/keymaps/vifon/keymap.o] Error 1
Make finished with errors
Makefile:535: recipe for target 'planck/rev6:vifon' failed
make: *** [planck/rev6:vifon] Error 1
```
Error is related to the use of `LAYOUT_planck_mit` macro:

`keyboards/planck/keymaps/vifon/keymap.c:36:13: error: implicit declaration of function 'LAYOUT_planck_mit' [-Werror=implicit-function-declaration]`

Any idea on what's happening?

fauxpark, 2019-06-09:
Yeah, this is a known issue that is being addressed in https://github.com/qmk/qmk_firmware/pull/4402.

drashna, 2019-06-17:
Theoretically resolved now. 

Re-open if that's not the case.

closed 2019-06-17 by drashna

joecabezas, 2019-06-17:
thanks everyone, will check tomorrow :)