qmk/qmk_firmware #6108: Tapping Toggle not working

ChrisNoob6460, 2019-06-09:
I have inserted #define TAPPING_TOGGLE 2 into my config.h file, and flashed with it, yet it still requires me to tap 5 times (default) to access my TT layer. I'm not sure if it's my defining problem, or that my config.h file isn't compiled into the hex file at all. One other possibility would be me having enabled retro-tapping, permissive hold, and others at the same time, not sure if it would clash with the definition. I do have my config.h file in my keymap folder during compiling, so I'm not sure where I went wrong, or if I did. Attached config.h below for reference:

```
#ifndef CONFIG_H
#define CONFIG_H

#include "config_common.h"

/* USB Device descriptor parameter */
#define VENDOR_ID       0xFEED
#define PRODUCT_ID      0x2260
#define DEVICE_VER      0x0001
#define MANUFACTURER    KBDFans
#define PRODUCT         DZ60
#define DESCRIPTION     DZ60 Keyboard

/* key matrix size */
#define MATRIX_ROWS 5
#define MATRIX_COLS 15

/* key matrix pins */
#define MATRIX_ROW_PINS { D0, D1, D2, D3, D5 }
#define MATRIX_COL_PINS { F0, F1, E6, C7, C6, B7, D4, B1, B0, B5, B4, D7, D6, B3, F4 }
#define UNUSED_PINS

/* COL2ROW or ROW2COL */
#define DIODE_DIRECTION COL2ROW

/* number of backlight levels */
#define BACKLIGHT_PIN B6
#define BACKLIGHT_LEVELS 5

/* Set 0 if debouncing isn't needed */
#define DEBOUNCING_DELAY 5

/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE

/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE

#define RGB_DI_PIN E2
#define RGBLIGHT_ANIMATIONS
#define RGBLED_NUM 16
#define RGBLIGHT_HUE_STEP 8
#define RGBLIGHT_SAT_STEP 8
#define RGBLIGHT_VAL_STEP 8

#define FORCE_NKRO
#define RETRO_TAPPING
#define TAPPING_TERM 150
#define TAPPING_TOGGLE 2
#define PERMISSIVE_HOLD
#define IGNORE_MOD_TAP_INTERRUPT

#endif

```

drashna, 2019-06-17:
Is the keyboard's config.h?  

if not, then that's the problem.  `#ifndef CONFIG_H` is wrong, and already defined. 

In fact, we've moved away from using that, and replacing:
```c
#ifndef CONFIG_H
#define CONFIG_H
```
with:
```c
#pragma once
```

ChrisNoob6460, 2019-06-17:
Thanks for the reply! Yes, what I listed above is a direct copy of my config.h. I just tried changing the config.h file and replacing what you listed to `#pragma once` , however i was unable to compile it succesfully, prompting an error with `#enddif ` . I tried just deleting `#enddif ` out of curiosity and was able to compile it succesfully, but after flashing I was unable to type anything at all except for toggling RGB and the soft reset button. I'm sorry for this really crude way of describing, as I am still new to QMK and coding in general, and that the Newbies guide didn't cover the specific details.

pepsiplease69, 2021-07-16:
do you mean `#enddif` or `#endif` 

seems like a typo to me

ksbrar, 2022-06-18:
bump, interested in resolution for this

pepsiplease69, 2022-06-18:
Can you please provide the command you’re using to perform the compilation, and please specify which folder you’re running this command from.

Please try the following:

- Comment out the define you made for the TAPPING_TOGGLE directive. Run the build command.
- Without making any changes, simply re-run the build command. This second time it should produce very little output because no code was changed and so none of the binaries have to be re-made
- Make the change in the config.h (uncomment you define statement). Recompile. This time around you should see a lot of output being produced because a lot of files are dependent on config.h and everything needs to be rebuilt after this code change. If at this point you’re not seeing much output being produced, then maybe you’re editing the wrong file.
- Possibly try to delete the hex file and force it to rebuild again.

The config.h you should be editing is the one that resides in your keyboard folder.
You will find another config.h file in your keymap folder, this file is not the correct one you need to edit. You must edit the one that is in your keyboard folder.

#define TAPPING_TOGGLE  2

I tried to make use of this directive and for me, everything worked as expected. 

I had my layer key configured to do MO(layer) and I redefined it to do TT(layer). 

I compiled everything and flashed my keyboard. Tap toggle worked as expected. The layer was enabled after 5 rapid taps of the key.

Then I made the change to my config.h and added the define statement. I then did a re-compile and this time I saw quite a lot of things being compiled. I flashed my keyboard afterwards.

Now my keyboard enables the layer after only two rapid taps of the key, just as expected.

Please let me know if you are able to address your issue with the above steps.

Thanks

pepsiplease69, 2022-06-18:
The command you should use to compile is the following, (please substitute your keyboard name and keymap names within the command)

qmk compile -kb [keyboard] -km [keymap]

It should be invoked from your top level qmk directory.

drashna, 2022-08-14:
I can't reproduce this, at all. 

I've had `#define TAPPING_TOGGLE 1` in my config for a long time now, and changing the value definitely changes the behavior appropriately.

BrknRobot, 2023-07-25:
I may have stumbled into this bug, or at least something similar. I've had TAPPING_TOGGLE set to 2 for years, but when syncing my layout with the last 2 years of QMK updates `TAPPING_TOGGLE 2` no longer works. `TAPPING_TOGGLE 1` (the default for my keyboard) works fine however. See https://github.com/BrknRobot/qmk_firmware/tree/npd/keyboards/ergodox_ez/keymaps/npd for the affected layout