qmk/qmk_firmware #6112: WhiteFox - RESET goes to "Kiibohd DFU Secure", cannot flash - needs to enter "Kiibohd DFU"

imchipwood, 2019-06-10:
## Describe the Bug
When using the RESET keycode in QMK firmware, the WhiteFox enters "Kiibohd DFU Secure" (as seen via dfu-util --list). In this state, it appears it cannot be flashed - dfu-util reports an error trying to copy data to the MCU. When using the reset button on the back of the keyboard, it enters "Kiibohd DFU" and is able to be flashed. See "Additional Context" section for output of dfu-util list and download commands in the two modes.

Essentially, the RESET keycode isn't useful for the WhiteFox (and presumably other boards using the Kiibohd bootloader). 

## System Information

 - Keyboard:  WhiteFox
   - Revision (if applicable):  True Fox
 - Operating system: Ubuntu 18.04
 - AVR GCC version: avr-gcc (GCC) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.6.398

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

#### Using QMK RESET keycode (flash fails):
$ sudo dfu-util --list
dfu-util 0.9
Found DFU: [1c11:b007] ver=0295, devnum=89, cfg=1, intf=0, path="1-5", alt=0, name="_**Kiibohd DFU Secure**_", serial="85C100007DEB00150007001032194E45 - mk20dx256vlh7"

$ sudo dfu-util -D .build/whitefox_imchipwood.bin 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [                         ]   0%            0 bytes
_**dfu-util: Error during download**_

#### Using reset button on back of keyboard (flash successful):
$ sudo dfu-util --list
dfu-util 0.9
Found DFU: [1c11:b007] ver=0295, devnum=91, cfg=1, intf=0, path="1-5", alt=0, name="_**Kiibohd DFU**_", serial="85C100007DEB00150007001032194E45 - mk20dx256vlh7"

$ sudo dfu-util -D .build/whitefox_imchipwood.bin 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [=========================] 100%        37440 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!

fauxpark, 2019-06-10:
From what I can tell, the Secure DFU mode is bypassed in three ways:
 - The physical reset switch is pressed
 - A watchdog timeout occurs
 - The MCU is unprogrammed

https://github.com/kiibohd/controller/blob/8c603cebed5608ada81f2f21dea4687613aa2f99/Bootloader/Devices/mk22fx512avlh12.c#L79-L85

It *may* be possible to do something with the watchdog, but I have no clue about these MCUs.

imchipwood, 2019-06-10:
Interesting finding, thanks! I'm not familiar with the watchdog subsystem, I'll try to dig around.

lzhoucs, 2020-03-09:
I just ran into the exact same issue on my whitefox too. My QMK firmware version is relatively new: `0.8.13`.

tzarc, 2020-03-09:
Is this a new thing, as in, it previously worked correctly? Or is this something that's been around for a while now?

lzhoucs, 2020-03-10:
Sorry I just recently switch to QMK so I don't know how it worked previously. But it sounds like it has been a problem since it was firstly reported.

americanhanko, 2021-01-24:
Possibly related to **Cannot enter DFU mode with whitefox's reset key** https://github.com/qmk/qmk_firmware/issues/4895

ToyKeeper, 2021-04-23:
Same issue here.  To reflash, I have to use a paperclip on the physical reset button under the keyboard.  QMK's reset key gets it to a state where the flashing tool detects it, but flashing fails.

Looks like bug #4895 is a dup.  Can it be closed?  It's older, but this newer one has more useful info.

Andrew-Fahmy, 2021-12-08:
I made a pull request with a fix for this issue, let me know if it works for you guys. https://github.com/qmk/qmk_firmware/pull/15430

closed 2021-12-08 by drashna

lzhoucs, 2022-11-24:
I just tried with latest QMK Firmware 0.18.17 and can confirm it does work now. However I still see "Kiibohd DFU Secure" instead of "Kiibohd DFU" although it doesn't matter anymore. Here is the full text between the qmk RESET and phsical reset button in ` dfu-util -l`:
* qmk RESET:
```
Found DFU: [1c11:b007] ver=0295, devnum=120, cfg=1, intf=0, path="3-4.2", alt=0, name="Kiibohd DFU Secure", serial="7EB100006F070036001C801331244D45 - mk20dx256vlh7"
```
* physical reset button:
```
Found DFU: [1c11:b007] ver=0295, devnum=122, cfg=1, intf=0, path="3-4.2", alt=0, name="Kiibohd DFU", serial="7EB100006F070036001C801331244D45 - mk20dx256vlh7"
```

PR #15430 (2021-12-08): Fix RESET not working for keyboards with Kiibohd bootloader

This is a fix for https://github.com/qmk/qmk_firmware/issues/6112. Keyboards with the Kiibohd bootloader would go into [secure DFU mode](https://github.com/kiibohd/controller/blob/master/Bootloader/Documentation/SecureDFU.md). In secure DFU mode the bootloader would do an extra key verification where it would look for a key in the first block of the new firmware. There is a way to disable this functionality, and after talking with @haata, I was able to implement this fix.

I have a K-type and this fixed the issue on my keyboard. If a [... cut]