qmk/qmk_firmware #6118: oneshot_mods_changed_user not called when key is licked

danielo515, 2019-06-11:
When you hit a one-shot modifier `ONESHOT_TAP_TOGGLE` times the key gets locked, and that is a useful feature that can also drive you mad if you activate it by accident.
I decided to use my Ergodox leds to signal the active one shot modifiers, and it worked quite well (code):

```js
// use leds to indicate when a one shot mod is on
void oneshot_mods_changed_user(uint8_t mods) {
  ergodox_board_led_off();
  ergodox_right_led_1_off();
  ergodox_right_led_2_off();
  ergodox_right_led_3_off();
  if (mods & MOD_MASK_SHIFT) {
    ergodox_right_led_1_on(); // One shot shift activated
  }
  if (mods & MOD_MASK_CTRL) { // oneshot control
    ergodox_right_led_2_on();  }
  if (mods & MOD_MASK_ALT) {// one shot alt
    ergodox_right_led_3_on();  }
  // if (mods & MOD_MASK_GUI) { // one shot gui
  //   println("Oneshot mods GUI");
  // }
  if (!mods) {
    rgblight_setrgb(250, 250, 80);
  }
}
```

But when the mod key gets locked none of them gets activated, and that is probably because oneshot_mods_changed_user is not being called when that happens.

Not sure if this is a bug report or a feature request (depends on the implementation intention 😄)

Regards

fauxpark, 2019-06-11:
I think you want to be using `oneshot_locked_mods_changed_user()` for that instead: https://docs.qmk.fm/#/feature_advanced_keycodes?id=callbacks

danielo515, 2019-06-12:
Is that callback executed for both scenarios or just when they got locked?

fauxpark, 2019-06-12:
As the name suggests, it's called only when the locked mods are changed, so you will have to use both in tandem.

drashna, 2019-06-12:
Though, for simplicity's take, you should be able to use this:

```c
void oneshot_locked_mods_changed_user(uint8_t mods) {
    oneshot_mods_changed_user(mods);
}
```

danielo515, 2019-06-13:
Thanks to both, but as usual, thank you very much @drashna for a great solution.

danielo515, 2019-06-13:
Thinking about it a bit further, will not them override their settings? Let me explain it:
If I lock alt, then the proper led gets on while all the others get off.
Then if I hit one shot control it will turn off all the leds and just turn on the control one, because the provided mods only includes changed ones... or does it include them all?
If the latter is the case, then this is completely solved.

Regards

danielo515, 2019-06-13:
Just tried and things works as I suspected.
Not sure how can I take advantage of both while keeping the values the other function sets up.

drashna, 2019-06-13:
A cheap hack would be to use `oneshot_mods_changed_user(mods|get_oneshot_mods())` instead.  

If I my brain is working, then that should merge the locked mods plus the oneshot mods, into one single set, and should get the effect that you want.

danielo515, 2019-06-13:
I tried your code @drashna and the result is exactly the same as just passing the same argument. When a one-shot modifier gets locked and then I activate any other one shot the first one led indicator gets cleared.

drashna, 2019-06-17:
dang.  :(

Not sure then, other than tracking it externally, and checking it that way.

closed 2022-08-28 by zvecr