qmk/qmk_firmware #6121: Planck rev6 make failure in Arch Linux

karma0, 2019-06-12:
Attempting `make planck/rev6:default` fails with the output described below.

## Planck rev6 make fails in Arch Linux

```
$ make planck/rev6:default
QMK Firmware 0.6.400
Making planck/rev6 with keymap default

arm-none-eabi-gcc (Arch Repository) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: tmk_core/common/chibios/bootloader.c                                                    In file included from ./lib/chibios/os/common/ext/CMSIS/ST/STM32F3xx/stm32f303xc.h:168,
                 from ./lib/chibios/os/common/ext/CMSIS/ST/STM32F3xx/stm32f3xx.h:153,
                 from ./lib/chibios/os/common/startup/ARMCMx/devices/STM32F3xx/cmparams.h:72,
                 from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70,
                 from ./lib/chibios/os/rt/include/ch.h:81,
                 from tmk_core/common/chibios/bootloader.c:3:
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function 'enter_bootloader_mode_if_requested':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:372: .build/obj_planck_rev6_default/common/chibios/bootloader.o] Error 1
Make finished with errors
make: *** [Makefile:553: planck/rev6:default] Error 1

```
## System Information

 - Keyboard:  Planck
   - Revision (if applicable): rev6 
 - Operating system: Arch Linux
 - AVR GCC version: 9.1.0
 - ARM GCC version: 9.1.0
 - QMK Firmware version: 0.6.398
 - Any keyboard related software installed? No

fauxpark, 2019-06-12:
You'll need to downgrade your ARM GCC version - 9.1.0 apparently has some problems. I believe 8.3.1 is good.

datNoko, 2019-06-12:
> You'll need to downgrade your ARM GCC version - 9.1.0 apparently has some problems. I believe 8.3.1 is good.

Thanks. Had the same issue with my DZ65, downgrading to 8.3.0-1 worked for me.

drashna, 2019-06-17:
Awesome, glad to hear it.

closed 2019-06-17 by drashna

karma0, 2019-06-17:
Does this mean that there are no future plans to support 9.1.0, or by proxy, vanilla installs of Arch Linux?Curiosity more than anything; I don't mind using Docker to build & load.

TheZoq2, 2019-06-20:
I also ran into this and it seems like a temporary fix is to add the environment variable `ALLOW_WARNINGS=yes`