qmk/qmk_firmware #6129: Errors if compiling with 's support

matj1, 2019-06-14:
## Describe the Bug

If I try to compile my layout with plover sup

## System Information

 - Keyboard: XD75RE
 - Operating system: Arch Linux
 - GCC version: 8.3.0
 - AVR GCC version: 9.1.0
 - QMK Firmware version: 0.6.32
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] Other: Plover

## Additional Context

This is printed if I try to `make` my layout as described above (I included only the parts with errors.):

```
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c                                 In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:365:5: error: 'const' attribute on function returning 'void' [-Werror=attributes]
  365 |     void CDC_Device_Event_Stub(void) ATTR_CONST;
      |     ^~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:367:10: error: 'EVENT_CDC_Device_LineEncodingChanged' alias between functions of incompatible types 'void(USB_ClassInfo_CDC_Device_t * const)' {aka 'void(struct <anonymous> * const)'} and 'void(void)' [-Werror=attribute-alias=]
  367 |     void EVENT_CDC_Device_LineEncodingChanged(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo)
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: aliased declaration here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:367:10: error: 'EVENT_CDC_Device_LineEncodingChanged' specifies less restrictive attribute than its target 'CDC_Device_Event_Stub': 'const' [-Werror=missing-attributes]
  367 |     void EVENT_CDC_Device_LineEncodingChanged(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo)
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: 'EVENT_CDC_Device_LineEncodingChanged' target declared here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:371:10: error: 'EVENT_CDC_Device_BreakSent' alias between functions of incompatible types 'void(USB_ClassInfo_CDC_Device_t * const,  const uint8_t)' {aka 'void(struct <anonymous> * const,  const unsigned char)'} and 'void(void)' [-Werror=attribute-alias=]
  371 |     void EVENT_CDC_Device_BreakSent(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: aliased declaration here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:371:10: error: 'EVENT_CDC_Device_BreakSent' specifies less restrictive attribute than its target 'CDC_Device_Event_Stub': 'const' [-Werror=missing-attributes]
  371 |     void EVENT_CDC_Device_BreakSent(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: 'EVENT_CDC_Device_BreakSent' target declared here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:369:10: error: 'EVENT_CDC_Device_ControLineStateChanged' alias between functions of incompatible types 'void(USB_ClassInfo_CDC_Device_t * const)' {aka 'void(struct <anonymous> * const)'} and 'void(void)' [-Werror=attribute-alias=]
  369 |     void EVENT_CDC_Device_ControLineStateChanged(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo)
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: aliased declaration here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.h:369:10: error: 'EVENT_CDC_Device_ControLineStateChanged' specifies less restrictive attribute than its target 'CDC_Device_Event_Stub': 'const' [-Werror=missing-attributes]
  369 |     void EVENT_CDC_Device_ControLineStateChanged(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo)
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:356:6: note: 'EVENT_CDC_Device_ControLineStateChanged' target declared here
  356 | void CDC_Device_Event_Stub(void)
      |      ^~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.c:359:1: fatal error: opening dependency file .build/obj_xd75_matj1/./lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.td: Permission denied
  359 | }
      | ^
cc1: all warnings being treated as errors
compilation terminated.
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:360: .build/obj_xd75_matj1/./lib/lufa/LUFA/Drivers/USB/Class/Device/CDCClassDevice.o] Chyba 1
make: *** [Makefile:549: xd75:matj1] Chyba 1
Make finished with errors
```

(Chyba means error in Czech.)

drashna, 2019-06-14:
You're on avr-gcc 9.1.0.  That's most likely the issue (we've seen a number of problems with it).

Try downgrading to 8.3 or 7.3.

drashna, 2019-06-17:
Have you tried compiling with the different version of avr-gcc yet?

matj1, 2019-06-17:
I tried to compile it with avr-gcc 8.3.0 and got the same errors.

fauxpark, 2019-06-17:
Looks like this problem: https://github.com/abcminiuser/lufa/issues/130

The fix was backported in #3510. Please make sure your repo is up to date :)

matj1, 2019-06-17:
I tried to compile it with avr-gcc 8.2.0, it failed with the same errors. Older versions of avr-gcc are not available by my package manager.

matj1, 2019-06-17:
How should I make sure my repo is ut to date? I did `git fetch upstream` and nothing was downloaded and the errors are still there.

drashna, 2019-07-03:
`git fetch` doesn't update the repo, it just updates the info. 

This link covers keeping your repo up to date: 
https://docs.qmk.fm/#/newbs_best_practices

drashna, 2019-07-03:
Also, you don't have to install avr-gcc from your package manager. I don't.  

Just make sure that you download the correct files, and add the bin folder to PATH on your system, and it will work.   I do this for ... MacOS, Linux and Windows, and it works great (all on 8.3.0)

ubone, 2019-09-08:
you could try an older qmk release if you don't need recent changes
qmk_firmware-0.6.450 builds for me, newest ones don't

drashna, 2019-09-26:
> you could try an older qmk release if you don't need recent changes
> qmk_firmware-0.6.450 builds for me, newest ones don't

To be blunt, this isn't a good solution.  It may get things working temporarily, but the issue doesn't actually get fixed. 

@matj1 were you able to get this issue fixed?

matj1, 2019-09-26:
Sorry @drashna, I forgot about this, I didn't try to solve it much since then. Also I had overwitten the `master` branch of my fork, which is against [afore mentioned best practices](#issuecomment-508058188) and probably is preventing me from updating my fork (merge conflicts). I will probably delete my fork having all my layouts backed up and fork the repository again.

noroadsleft, 2019-09-26:
@matj1,

Whatever the state of your repo, if you have QMK's repo set as `upstream`, and your fork as `origin` as described in the Git Best Practices doc, you don't need to delete your fork entirely (and even if you don't, it's fixable). Checkout your local master branch (the one on your computer's hard drive/SSD), and run:

```
git checkout -b old_master
git push -u origin old_master
```

This creates a new branch that is a duplicate of your current master branch, and pushes it to GitHub, so you will still have all your extra commits.

From here, you can run:

```
git checkout master
git fetch upstream master
git reset --hard upstream/master
git push --force origin master
```

This will re-synchronize your master branch with QMK's master branch, overwriting any changes that aren't in QMK's master branch. You won't need to worry about losing your work, because it's still on the `old_master` branch. From here, you can `git checkout old_master` and resume working. Additional changes can all be done here, and you'll have a clean master branch for testing against QMK's master when the time comes to do that.

closed 2022-08-03 by tzarc