qmk/qmk_firmware #6147: QMK Toolbox crashes on MacOS Catalina when trying to flash dz60rgb

sidcarter, 2019-06-18:
When trying to flash a DZ60RGB using the flash or the auto flash option causes QMK Toolbox to crash on MacOS Catalina Beta 1 (10.15)

## Describe the Bug
Created a custom keymap and was trying to flash my dz60rgb using QMK Toolbox.

When I set QMK Toolbox to auto flash or manually try to flash it, QMK Toolbox crashes. I've also got the latest Xcode beta installed, if that matters.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
      DZ60RGB 
 - Operating system:
     Mac OS Catalina Beta 1 (10.15)
 - AVR GCC version: 
```
avr-gcc --version
avr-gcc (GCC) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

 - ARM GCC version: 

 - QMK Firmware version:
```
git describe --abbrev=0 --tags
0.6.379
```

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Details from the crash reporter:
```
Process:               QMK Toolbox [34384]
Path:                  /Applications/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
Identifier:            fm.qmk.toolbox
Version:               0.0.10 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QMK Toolbox [34384]
User ID:               502

Date/Time:             2019-06-17 21:11:06.124 -0400
OS Version:            Mac OS X 10.15 (19A471t)
Report Version:        12
Bridge OS Version:     3.0 (14Y903)
Anonymous UUID:        D11A708D-7E35-C0B1-498A-E89904D97849

Sleep/Wake UUID:       0F1403D0-7012-4F6A-B01D-73831943AAD8

Time Awake Since Boot: 54000 seconds
Time Since Wake:       4000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'launch path not accessible'
[... 422 lines cut ...]
__LINKEDIT                       346.4M        7 
__OBJC_RO                         30.3M        1 
__OBJC_RW                         1652K        2 
__TEXT                           185.2M      283 
__UNICODE                          564K        1 
mapped file                       88.1M       28 
shared memory                      672K       21 
===========                     =======  ======= 
TOTAL                              1.3G      832 
TOTAL, minus reserved VM space   984.8M      832 

Model: MacBookPro14,3, BootROM 194.0.0.0.0, 4 processors, Intel Core i7, 2.8 GHz, 16 GB, SMC 2.45f0
Graphics: kHW_IntelHDGraphics630Item, Intel HD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro555Item, Radeon Pro 555, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E434C47414C41522D4E5644
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E434C47414C41522D4E5644
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x173), Broadcom BCM43xx 1.0 (7.77.87.1 AirPortDriverBrcmNIC-1420.2)
Bluetooth: Version 7.0.0d88, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: STM32  BOOTLOADER
USB Device: iBridge
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1
```

fauxpark, 2019-06-18:
Duplicate of https://github.com/qmk/qmk_toolbox/issues/84

sidcarter, 2019-06-18:
oh... chmod+x of dfu-util fixed the problem here too. Cool!

closed 2019-06-18 by sidcarter