qmk/qmk_firmware #6151: Planck EZ won't build

pjanx, 2019-06-18:
Planck EZ firmware will not build, see command line invocation below.

Might be that it should be built some other way, in which case documentation is desired.

Freshly git-pulled. Thanks in advance.

```
$ CFLAGS="-Wno-error=deprecated" make -j1 planck/ez:default:all                                                                     master
QMK Firmware 0.6.400
Making planck/ez with keymap default and target all

arm-none-eabi-gcc (Arch Repository) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/planck/planck.c                                                               In file included from keyboards/planck/planck.c:1:
keyboards/planck/planck.h:145: error: "KEYMAP" redefined [-Werror]
  145 | #define KEYMAP LAYOUT_ortho_4x12
      | 
In file included from keyboards/planck/planck.h:9,
                 from keyboards/planck/planck.c:1:
keyboards/planck/ez/ez.h:105: note: this is the location of the previous definition
  105 | #define KEYMAP LAYOUT_planck_grid
      | 
In file included from keyboards/planck/planck.c:1:
keyboards/planck/planck.h:147: error: "LAYOUT_planck_grid" redefined [-Werror]
  147 | #define LAYOUT_planck_grid LAYOUT_ortho_4x12
      | 
In file included from keyboards/planck/planck.h:9,
                 from keyboards/planck/planck.c:1:
keyboards/planck/ez/ez.h:88: note: this is the location of the previous definition
   88 | #define LAYOUT_planck_grid( \
      | 
keyboards/planck/ez/ez.h:106:27: error: implicit declaration of function 'LAYOUT_planck_grid' [-Werror=implicit-function-declaration]
  106 | #define LAYOUT_ortho_4x12 LAYOUT_planck_grid
      |                           ^~~~~~~~~~~~~~~~~~
keyboards/planck/planck.h:147:28: note: in expansion of macro 'LAYOUT_ortho_4x12'
  147 | #define LAYOUT_planck_grid LAYOUT_ortho_4x12
      |                            ^~~~~~~~~~~~~~~~~
keyboards/planck/planck.c:24:53: note: in expansion of macro 'LAYOUT_planck_grid'
   24 | const uint8_t music_map[MATRIX_ROWS][MATRIX_COLS] = LAYOUT_planck_grid(
      |                                                     ^~~~~~~~~~~~~~~~~~
keyboards/planck/ez/ez.h:106:27: error: invalid initializer
  106 | #define LAYOUT_ortho_4x12 LAYOUT_planck_grid
      |                           ^~~~~~~~~~~~~~~~~~
keyboards/planck/planck.h:147:28: note: in expansion of macro 'LAYOUT_ortho_4x12'
  147 | #define LAYOUT_planck_grid LAYOUT_ortho_4x12
      |                            ^~~~~~~~~~~~~~~~~
keyboards/planck/planck.c:24:53: note: in expansion of macro 'LAYOUT_planck_grid'
   24 | const uint8_t music_map[MATRIX_ROWS][MATRIX_COLS] = LAYOUT_planck_grid(
      |                                                     ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:372: .build/obj_planck_ez_default/keyboards/planck/planck.o] Error 1
Make finished with errors
make: *** [Makefile:553: planck/ez:default:all] Error 1
```

## System Information

 - Keyboard:  
   - Revision (if applicable): 
 - Operating system: Archlinux
 - AVR GCC version: 9.1.0
 - ARM GCC version: 9.1.0
 - QMK Firmware version: 0.6.400
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

fauxpark, 2019-06-18:
Yeah, the Planck EZ is broken at the moment. I think you can get around this for now by removing the layout defines in `ez.h` - the issue relates to them being already defined in `planck.h`.

I should also note that you may have problems compiling with AVR/ARM GCC version 9.1.0. There have been reports that it's broken too, so you might have to downgrade to 8.3.0.

drashna, 2019-06-18:
^  This.  We're working on getting it fixed in a non temporary way. 

Also, you can run `git revert f4840139a` until this has been fixed.

pjanx, 2019-06-18:
Just for completenes, there is also https://configure.ergodox-ez.com/planck-ez/layouts/default/latest/0 which I've used and covers most use cases.

drashna, 2019-06-26:
This should be fixed now.

closed 2019-07-03 by drashna