qmk/qmk_firmware #6152: Symbol keys switch in Iso osx

aykot, 2019-06-18:
## Describe the Bug

As soon as I press º key, system recognizes > and they are both switched until I reset the keyboard. At first I thought it was a problem with my first build ( and adafruit blufeather) but I have the same problem now with a pub ymd96. Tested in Mojave, Yosemite and high Sierra. Same output

## System Information

 - Keyboard:  
   - Revision (if applicable):  ymd96
 - Operating system: osx Mojave
 - AVR GCC version: 9.1.0

 - ARM GCC version: 7.3.1

 - QMK Firmware version:0.6.398

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Present at least from qmk gh60-0.1

fauxpark, 2019-06-18:
If you go to the Keyboard preferences -> Input Sources, does the graphic there show an ISO layout?

aykot, 2019-06-18:
yes @fauxpark , and when I press the º button, it switches the keyboard map shown in that picture

fauxpark, 2019-06-18:
Which input source are you using?

aykot, 2019-06-18:
I´ve tried both Spanish and Spanish - ISO and both have the same problem

fauxpark, 2019-06-18:
Could you show a picture?

fauxpark, 2019-06-18:
Please also post the keymap you're using on the YMD96.

aykot, 2019-06-18:
[video.zip](https://github.com/qmk/qmk_firmware/files/3302558/video.zip)

[me96.json.zip](https://github.com/qmk/qmk_firmware/files/3302560/me96.json.zip)

fauxpark, 2019-06-18:
Hm, it looks like you've generated that .json file using kbfirmware, which is not recommended as it's not an official tool and very out of date. That might be (part of) the issue. Please have a look at the [QMK Configurator](https://config.qmk.fm) instead.

aykot, 2019-06-18:
I have the same problem with /ymd96/keymaps/AnthonyWharton keymap, which is generated with qmk, and with my other keymap in a complete different build

fauxpark, 2019-06-18:
Oh, I think I see now. What you want to do is make sure the layout has changed to the ANSI one (as you showed in the video), then click the `Change keyboard type...` button in the Keyboard preferences and set it up as an ISO layout.

aykot, 2019-06-18:
I think you are right and the problem goes in that direction, but Mac doesn't recognices qmk keyboard as an unknown keyboard, so there is no "change keyboard type" and defaults it to ANSI, without possibility to change it, even when deleting the corresponding plist. Is there any part in qmk where I can specify that it is ISO instead of ANSI?

fauxpark, 2019-06-18:
It's entirely on the OS side, QMK can't do anything about it. Try running the following:

```
sudo open /System/Library/CoreServices/KeyboardSetupAssistant.app/Contents/MacOS/KeyboardSetupAssistant
```

aykot, 2019-06-18:
ok, found the problem and fixed it.
Osx recognizes the keyboard as ANSI by default, but if you delete /Library/Preferences/com.apple.keyboardtype.plist and then restart the system, it prompts with a window where you can select keyboard type.
Now my system recognizes it as follows:

[image] 

Thank you @fauxpark

closed 2019-06-18 by aykot