qmk/qmk_firmware #6160: Improve marketing on proton-c page

danielo515, 2019-06-19:
This is the headline for the proton-c:

> The Proton C is an ARM STM32F303xC based drop-in replacement for the Pro Micro. It uses the same chip as the Planck rev6 boards, and Skull’s new ARM Clueboards.

While everything in there is probably correct (I didn't checked but I trust you) it says nothing to the average user. Even for mid users like me (I don't consider myself an advanced user, but I have several keyboards and I have even built my own from scratch) it does not tells me what I want to know. This is what I want to read first when I open that page:

- Does it have more memory? the answer seems to be YES! This is key!
- Does it have a faster processor? Again seems that the answer is YES!

And just with those two bullet points I'm sold. But currently, after reading that page I just feel: should I consider using this? Why?
I think we should make clearer that it is better and why it is better.

Regards

drashna, 2019-06-19:
Well, to the point, a good comparison may be best. 

---- | ATMega32u4        | STM32F303xC
-------------- | ------------------- | ----------------
Speed       | 16MHz            | 72MHz
Voltage    | 5v                      | 3.3v with some 5v capable pins
Flash size | 32KB (28 usable) | 256KB usable
Memory   | 2.5KB               | 48KB
Timers      | 4 total              | 10 total
Digital IO | 25 pins             | 37 pins

And more.  But I think this covers the important stuff.

noroadsleft, 2019-06-19:
Good spot, @danielo515. If you'd like to improve the page, the file that needs updating is:

https://github.com/qmk/qmk.fm/blob/gh-pages/_pages/proton-c.md

danielo515, 2019-06-20:
Wow @drashna, nice comparison table!
I like it. The only thing I see we can improve is adding a sentence in plain English saying how much more memory for your firmware you have (better in %). At first, I thought it was 32KB VS 256KB, but then the EEPROM thing confused me. Does that mean that the extra memory on the proton-c is `256KB - 32KB - 1024b` ?

The memory row I guess it is about the "ram" (probably not technically ram but, whatever)

noroadsleft, 2019-06-20:
I'm not 100% on how the EEPROM works on STM32 chips, but yes, the 256kB is the flash memory - the EEPROM data also lives here - and the 48kB is the RAM.

drashna, 2019-07-03:
The EEPROM uses program space, so eats into that 256KB size.   However, there is no actual limit on EEPROM size, and QMK Firmware will continue to allocate and use however much space is set for EEPROM.  This means that we could actually store all 32 layers in eeprom (for dynamic keymap support, for instance), without any issues.   And IIRC, it starts at the end, and works it's way backwards.  But @yiancar would need to confirm what, since they're the expert on this. 

---- | ATMega32u4        | AT90USB1286 | STM32F303xC
-------------- | ------------------- | ---------------- | ----------------
Speed       | 16MHz            | 16MHz        | 72MHz
Voltage    | 5v                      | 5v                | 3.3v with some 5v capable pins
Flash size | 32KB (28 usable) | 128KB (120KB usable) | 256KB (all usable)
EEPROM size | 1024 bytes | 4098 Bytes |emulated on flash storage
Memory   | 2.5KB               | 8KB              |48KB
Timers      | 4 total              | 4 total         |10 total
Digital IO | 25 pins             | 46 pins        | 37 pins

The STM32F303 has 9 times the usable storage space that the ATMega32U4 has.   You could enable a bunch of features and still have plenty of room  left.  For instance, my Planck EZ, which has audio, and rgb matrix enabled with no sub-features disabled, and has some of the compiler optimizations turned off uses 45KB of firmware, leaving 200+KB free, still.   Plenty of room for most anything you want to do.

yiancar, 2019-07-03:
The eeprom space will not dynamically expand. The pages need to be reserved and never changed for that keyboard. If the eeprom settings change the pages will probably be formated. 

Also if you start eating too much into the flash you need to start letting the compiler know by modifying the .ld life. Right now the space used is so far (last 4 pages) so we just assume and hope that the compiler will not put anything there.

drashna, 2019-07-13:
@yiancar how do you set that?  The pages reserved?

yiancar, 2019-07-15:
https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/chibios/eeprom_stm32.h#L43

FEE_DENSITY_PAGES sets the number of reserved pages.

Here you need to remember that you need double the number of pages than what you are going to use.
This is because you always need a page fre to be able to set as "dirty"

drashna, 2019-07-15:
So if I'm reading that right, the `MCU_STM32F303CC` uses 4KB of eeprom?

yiancar, 2019-07-15:
So, there is 4 pages reserved. For STM32F303, each page is 2KB, therefor 8KB is reserved for eeprom, of which 4KB are usable.

drashna, 2019-07-15:
Wooo, I can do maths! :D

closed 2019-07-23 by drashna