qmk/qmk_firmware #6168: LED indicator for scrolllock not showing

numv, 2019-06-20:
after flashing massdrop/ctrl:default on my board the scrolllock active led indication is not working and the caps indication is kinda working but behaving strangely.

## System Information

 - Keyboard:  massdrop ctrl
 - Operating system: windows wsl

## Additional Context

if i press capslock i see a rainbow effect that distincts the key from the rest of the rainbow effect
If i press scroll lock i dont see any indication that it is active.

now if i press FN=Z the RGB_TOG toggles so all modkeys and the underglow turns off.
if i now press the scroll lock it flickers white. if i press it again it sometimes turns still white and sometimes turns off. If i press the caps key it flickers white if i press it again it glows white then toggles between glowing white and flickering if you hit it multiple times it will turn off.

numv, 2019-06-20:
if you don't understand what i am talking about:
https://www.youtube.com/watch?v=D4Ab1dz_aWo

drashna, 2019-07-03:
Honestly, looking at the current code for the CTRL in the repo, I don't see it indicating this, at all. 

However, you could simply add this.  To do so, add: 
```c 
void rgb_matrix_indicators_user(void) {
    if (IS_LED_HOST_ON(USB_LED_CAPS_LOCK)) {
        rgb_matrix_set_color( 51, r, g, b);
    }
    if (IS_LED_HOST_ON(USB_LED_SCROLL_LOCK)) {
        rgb_matrix_set_color( 15, r, g, b);
    }
}
```

I'm not 100% certain on these LEDs numbers, so you may need to tweak them.  But that should get you close to what you want.

numv, 2019-07-03:
I too did not found anything for the indicator.
But it must be somewhere because it kinda works.

I did it with something like you said, but thats kinda hacky:
```javascript
void rgb_matrix_indicators_user(void) {
    if (this_led & (1 << USB_LED_SCROLL_LOCK)) {
        rgb_matrix_set_color(14, 0xFF, 0x00, 0x00);
    }
}
```
beacause the led indication doesn't reflect the current running effects
for example if you are on a static color or if it is on the rainbow effect like  in the Yt video

numv, 2019-07-03:
>         rgb_matrix_set_color( 15, r, g, b);

Does it work to just type `r, g, b` or did you just used them as a placeholder?

drashna, 2019-07-03:
No, that's perfectly fine!   It's not hacky.  In fact, the check I used above basically does the same thing.  

https://github.com/qmk/qmk_firmware/blob/0f8431a57f4b1ce50528e4c1689e810ba9554e17/tmk_core/common/host.h#L25-L29

As for the `r, g, b`, no, you'd need to add the specific colors that you want to use.   I just used them as a placeholder.

Also, there is a list of predefined colors here: 
https://github.com/qmk/qmk_firmware/blob/22ba36a4d86c06ce1b6049555525646eb16bb57a/quantum/rgblight_list.h#L19-L36

numv, 2019-07-03:
Yeah but i meand hacky in the way that you wouldn't mind the current rgb effect.
If you look at the CAPSLock indication in the Youtube Video.
It reflects the rainbow effect.

If i use `rgb_matrix_set_color` and have the rainbow effect active and press scroll lock it will be a static color for example red, (well its not bad but the rainbow on the single key would be pretty neat) and if i press it again the rainbow will continue.
Now the problem is when you use the `rgb_matrix_set_color` function and the static color effect.
Your led indicator wont change back to the effect color like it was before.
You could probably do it yourself check which color the indicator led currently is lighting as, then toggle it to an defined active color and back if you deactivate scrolllock.
But why is it working on the CAPSLock without doing anything ;P.
Is it the massdrop keyboard itself doing that?

numv, 2019-07-03:
Also the `rgb_matrix_set_color` doesn't work well with toggling the rgb effect.
You can set the LED even if it should be off via (i would need to check it myself in the keymap file).

https://github.com/qmk/qmk_firmware/blob/c0c8b2e50d0da5813aac3c6bc69540f81b1dae04/keyboards/massdrop/ctrl/keymaps/default/keymap.c#L105

yungkneez, 2020-04-04:
Glad I'm not the only one experiencing this. It seems to depend on the effect you have chosen. If I have the RGB_MATRIX_SOLID_REACTIVE_SIMPLE effect set, my caps lock indicator works fine but the scroll lock indicator behaves like any other key, thus making it impossible to tell if it is on or off.

nblyumberg, 2020-05-14:
I tested this on my CTRL board with the default keymap and I also see this  behavior.

nblyumberg, 2020-05-17:
In case anyone else comes across this, this is not a keymap issue and something either with the specific massdrop microcontroller code or something simiar as these will all not work with scroll lock:

- known good code in a custom keymap to control caps lock, does not work with scroll lock
- using the default massdrop firmware (no scroll lock)
- using the default keymap in the QMK repo
- using other keymaps in the QMK repo

ewersp, 2020-06-14:
Hey, so I was seeing this too on my Drop ALT with the default QMK firmware, and managed to implement a fix that can be applied in your `keymap.c`:

```
// Taken from 'g_led_config' in config_led.c
#define CAPS_LOCK_LED_ID 30

// This runs every matrix scan (every 'frame')
void rgb_matrix_indicators_user(void) {
    led_flags_t flags = rgb_matrix_get_flags();

    // If we're in either keylight or underglow modes (but not both simultaneously)
    if (HAS_FLAGS(flags, LED_FLAG_KEYLIGHT) != HAS_FLAGS(flags, LED_FLAG_UNDERGLOW)) {

        // This fixes a bug where the caps lock LED will flicker when toggled in either keylight or underglow modes
        if (host_keyboard_leds() & (1 << USB_LED_CAPS_LOCK)) {
            rgb_matrix_set_color(CAPS_LOCK_LED_ID, 0xFF, 0xFF, 0xFF);
        } else {
            rgb_matrix_set_color(CAPS_LOCK_LED_ID, 0x00, 0x00, 0x00);
        }
    }
}
```

I set the color to white by default (0xFFFFFF) but you can set it however you'd like. The underlying problem was only in underglow or keylight modes, where the caps lock LED is off by default. The 'indicator' code to override the LED was working correctly in` led_matrix.c led_matrix_indicators()`, but there was no code to actually reset it after disabling an indicator key (eg: caps lock), so it was constantly inverting the color from 0 -> 255 -> 0 -> 255 -> etc. very quickly causing the flicker. If you just so happened to disable caps lock during the 'frame' when it was 0, the light would stay off.

I do think there is an underlying issue here, but I couldn't find a general solution.

Initially I tried storing the previous keyboard state in led_matrix.c, eg: `uint8_t prev_kbled = keyboard_leds();` and modified `led_matrix_indicators()` to check when an indicator key was released, and if so, reset the LED back to 0x000000, but the code got a bit messy so I went with an alternate approach. This did work though.

Here's a video I recorded after the fix: https://imgur.com/gallery/MFkjWjI
And this video was recorded by someone else showing the original bug: https://imgur.com/gallery/ZmJ8lWh

Hope this helps!

closed 2022-05-08 by numv