qmk/qmk_firmware #6170: Random characters sent sometimes (gergo)

leshow, 2019-06-21:
It's hard to describe, while typing sometimes the keyboard will completely spaz out and start sending a bunch of random characters, this has happened on multiple machines and over multiple versions of QMK, but always on the gergo keyboard.

My layout isn't particularly complicated and can be found [here](https://github.com/leshow/qmk_firmware/blob/master/keyboards/gergo/keymaps/leshowmap/keymap.c)

I originally thought it was due to spacecadet shift but after replacing with regular shift, it's still happening.

 - Keyboard:  Gergo
 - Revision (if applicable):  Rev 02: Distortion
 - Operating system: Linux (Debian & Arch)
 - AVR GCC version: 9.1.0
 - QMK Firmware version: 0.6.393
 - Any keyboard related software installed? 
    None

drashna, 2019-06-21:
Well, could you try using avr-gcc 8.3.0?  

We've seen issues with 9.1.0, and there may be problems with the compiler.

leshow, 2019-06-21:
I could, but I just edited my post to say I also have an ergo infinity compiled with a similar layout that doesn't seem to have any problems.

Come to think of it, my debian machine has got to have an older version of avr also.

leshow, 2019-06-21:
Yup, sorry I didn't word that sentence properly. I will absolutely try
downgrading and follow up.

leshow, 2019-06-21:
My debian box is running an ancient avr-gcc 4.9.2 and produces the same behaviour (for the gergo, but not the ergodox_infinity).

leshow, 2019-06-22:
Okay, so I've downgraded to 8.3. I've also tried using the configurator at https://config.qmk.fm/ I basically re-made my layout in the configurator and downloaded the compiled hex file & flashed it with `dfu-programmer`.

Nothing managed to alleviate the issue. Is it possible I've got a bad cable or something? The gergo is split in 2 halves connected with a trrs cable

drashna, 2019-06-22:
Okay, then it sounds like it may be an issue with the matrix stuff, or it may be an issue with the board itself. 

Unfortunately, it's not exactly an easy fix. :(

leshow, 2019-06-23:
I don't know much about electronics what are the possibilities with behavior like this? It seems like the junk input is mostly coming from the left half of the board.

drashna, 2019-07-03:
More than likely, it's an issue with the matrix code.  Unfortunately, it's older code and may need some updating. 

Also, comparisons to the ergodox infinity aren't really helpful, as it's a completely different board (AVR vs ARM, and port expander vs dual controllers).

ieure, 2019-07-05:
I've run into this on two separate Ergodox Infinity boards.  I never had issues with my classic Ergodox.

I'll have to keep better track next time this happens, but when it occurs, it seems to spit out several (6ish) characters which look similar to hex.  Unplugging / replugging fixes it.

It's uncommon, maybe once in a month of daily 8 hour / day use, but very annoying.

leshow, 2019-07-07:
That's odd, it happened with me every 20 mins or so, making the board
basically unusable. I put the board together myself and have never SMD
soldered before, so I can't rule out that I messed something up with
assembly. I feel like the number of things I could've done to cause a
problem like this is pretty small though. I will be getting a gergo already
assembled and if it starts sending junk input or I'll be contacting the
seller.

ieure, 2019-08-29:
Okay, I've had this happen a few times over the last few days, and since I'm on a laptop, I was able to use the built-in keyboard to capture some of what's going on.

This happened when I tried to type an ampersand.

Here are all the alpha keys and what the keyboard outputs when I type them:

```
q
w
e
r 0708 0708 0708
t 1eb1 1eb1 1eb1
y 26bc 26bc 26bc
u A A A
i
o 5f55 5f55 5f55
p
a 09d3 09d3 09d3
s
d 7762 7762 7762
f 7268 7268 7268
g d d d
h
j 580d 580d 580d
k
l 2867 2867 2867
z
x
c
v
b 1344 1344 1344
n
m
, a a a
. 0eef 0eef 0eef
```

Any key without text next to it either did nothing or output a non-printable character, like `<XF86Calculator>`, `<C-s-XF86Launch8>`, `<XF86Launch7>`, `<XF86Launch7>`, `<C-s-delete>`, etc.

[This is my current keymap](https://github.com/ieure/qmk_firmware/blob/master/layouts/community/ergodox/ime/keymap.c), which I built on 2019-07-05 and flashed to the keyboard the same day.  I have the .hex files as well, if those are helpful.

My shift keys have one-shot modifiers and are right next to two momentary layer switching keys.  Maybe I'm hitting a layer key by accident and that's causing the issue?

toshism, 2019-11-25:
I recently started having this issue as well. I'll just add that it's new for me. I had no problems and made no updates to my firmware from around the end of June 2019 until this month. After updating (and updating qmk at that point as well) I started having the issue described above. It could be a coincidence, but maybe it's useful info.

drashna, 2019-12-01:
Have you tried upping the debounce? 

Also, do you know which specific commit that caused the issue to start happening?

toshism, 2019-12-01:
This is not a bouncing type issue. It will continue to spew random characters even if i’m pressing no keys at all until i unplug the board. It seems to be triggered when hitting multiple keys at once I think, but I can’t even guarantee that.  Just quickly unplugging and replugging it back in “fixes” it though. 

I’m happy to help debug if someone can point me in the right direction on where to start.

ieure, 2019-12-02:
> It will continue to spew random characters even if i’m pressing no keys at all until i unplug the board.

This is not the behavior I'm experiencing.  My keyboard doesn't spew characters on its own, pressing a key produces consistent, wrong output.  ex. when I press "d", it outputs "7762".

yanfali, 2019-12-03:
@toshism if you are convinced it's not a hardware issue, I would bisect qmk source code trying to find the commit that started causing the issue. It's quite possible another change has triggered this bug. The other thing to check for is cold joints in your soldering and looking for shorts on your PCB. I would also have a chat with @germ as he is the designer of your keyboard.

toshism, 2019-12-05:
It seems like I am experiencing a different issue than described here. Sorry about confounding the two.

I would not say I am convinced it is not a hardware issue, I just suspected it could be software due to it starting after an update. That could very well be coincidence though and I will check my solder joints etc.

hermesjpappas, 2021-03-22:
Just got a Sofle board and this happening to me. Spewing characters on its own at random intervals. Tried unplugging and re-plugging and it didn't work. I'm wondering what is wrong, as I didn't build this board myself but the seller told me he had tested it. Could it be purely software-related? I have only flashed it using QMK configurator and then the QMK toolbox on Windows. Maybe I should try the compiling from Linux route?

avitkauskas, 2021-08-28:
It's an old issue, but still open, so I'll comment.
I had the same issue when keyboard just started bursting random characters at random intervals after the update of the firmwaren/keymap. The problem dissapeard when I reflashed both sides of the keyboard with the same firmware.

ieure, 2021-08-28:
This issue is still happening to me.  After reflashing my classic Ergodox, it's happening there as well as on the two Infinity Ergodoxes I mentioned before.  Seems like definitely a firmware issue, if it's happening across multiple examples of different keyboards.

I reflashed my daily driver about three weeks back, with 0.13.32, and I can still trivially reproduce the problem.

JesseCSlater, 2022-02-25:
I am having this same issue with the gergo. Sounds like the same symptoms. It could be something with the board? It just starts to spew a bunch of random key strokes, which activates keyboard shortcuts, and has closed my tab, and once even shut down my computer. It does not happen all the time, it has only happened to me on a few different days, but when it does happen, it keeps going for several minutes. Unplugging and re plugging fixes it temporarily, but then it comes back. After a while it goes away again, and then does not come back for a few weeks. I am hoping it does not become more regular, because if it does I will have to find a new ergo board to buy.

bromanbro, 2022-08-16:
I am experiencing this issue with an ID75 board.  Firmware was build online in QMK configurator.  This only happens once every couple of months.

ameyp, 2023-02-14:
This just started happening to me today on a `kinesis/kint` keyboard

aaannestad, 2024-03-16:
Add me to the pile of people with this issue; has now happened a couple of times on my Ergodox. Unplugging the board doesn't even seem to fix the issue immediately; it's not clear how to even reliably get the problem to stop temporarily.