qmk/qmk_firmware #6175: Can I just combine two key to one key?

kls1991, 2019-06-22:
I want a key to mean both MO(layer) and F13.
F13 is always useless so it will not conflict. When I need, it may do some tricks, like Pan in photoshop (hold one key to move the picture, LT cannt do that), like confirm in inputmethod, make mousekey to function more with AHK, etc......
So this is my idea, but I'm not a programmer I dont konw how to do this......
I kown I should not  ask for help here, but I am really not sure how to do this. I have read docs.qmk.fm for sevral hours, still not sure what is the best way to do this. Macro maybe?
Just a hint is enough, help please.
Sorry for my bad English.

noroadsleft, 2019-06-23:
Are you wanting the key to be F13 when it's tapped and `MO(layer)` when it's being held? If so, you can use `LT(layer, KC_F13)`, substituting the number of your destination layer for `layer`. This works the same as `MO(layer)` but adds the ability to send a different keycode if you just tap the key instead of hold it.

And this is a fine place to ask for help.

kls1991, 2019-06-23:
> Are you wanting the key to be F13 when it's tapped and `MO(layer)` when it's being held? If so, you can use `LT(layer, KC_F13)`, substituting the number of your destination layer for `layer`. This [...]

Thanks
I don't want LT() .
There will be a delay and I can't hold KC.
I want them both, just press F13 and MO simultaneously

noroadsleft, 2019-06-23:
Oh, so you want one key that sends a keydown event for F13 *and* changes the layer, is that correct?

kls1991, 2019-06-23:
> Oh, so you want one key that sends a keydown event for F13 _and_ changes the layer, is that correct?

Yes...

noroadsleft, 2019-06-23:
K, you'll need to make some changes directly to your `keymap.c` file.

## Step 1: Create the keycode to handle this functionality

```c
enum custom_keycodes {
    LYR_F13 = SAFE_RANGE
};
```

If you have custom keycodes defined already, you can add `LYR_F13` to the end of your existing `custom_keycodes` block:

```c
enum custom_keycodes {
    MACRO1 = SAFE_RANGE,
    MACRO2,
    LYR_F13
};
```

## Step 2: Write the keycode's functionality

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LYR_F13:
            if (record->event.pressed) {
                // when keycode LYR_F13 is pressed
                register_code(KC_F13);  // press the F13 key
                layer_on(_FN);          // turn on the _FN layer
            } else {
                // when keycode LYR_F13 is released
                layer_off(_FN);           // turn off the _FN layer
                unregister_code(KC_F13);  // release the F13 key
            }
            break;
    }
    return true;
}
```

`_FN` here can be whatever layer you wish to turn on when the key is pressed and off when released. If it doesn't have an defined alias like `_FN` or `_ADJUST`, you can use the layer's index number (like `layer_on(2)` or whatever).

## Step 3: Assign the new keycode in your keymap

```c
    [_BASE] = LAYOUT(
        KC_ESC,  LYR_F13, KC_F14,  KC_F15,  ...
        ...
    ),
    [_FN] = LAYOUT(
        ...
    )
```

----

This should get you going. Best of luck!

kls1991, 2019-06-23:
Thank you man !!!

noroadsleft, 2019-06-23:
You're welcome. Please let me know if you have any problems with it.

drashna, 2019-07-03:
As this is pretty much solved, I'm closing this.

closed 2019-07-03 by drashna