qmk/qmk_firmware #6182: New split firmware: keys don't register

LouWii, 2019-06-24:
I'm writing a new firmware for my custom keyboard. It's a split with a ProMicro on each side.

I was able to create a firmware for each side separately, which works great. But now I'm trying to make the firmware for a "real" split where the right side connects to the left side via I2C (TRRS cable), and the left side is the master side.

The problem is, there's not much documentation about how to configure QMK for a split? The only thing I found is https://docs.qmk.fm/#/config_options?id=split-keyboard-options I'm trying to get information from existing split keyboards but most of them use serial or are based on older versions of QMK and not the "new" split common code.

This is my code: https://github.com/LouWii/qmk_firmware/tree/master/keyboards/ergomax

I believe I didn't miss anything. But again, couldn't find any doc so hard to tell.

The problem is: no key tap are registered at all, with either the 2 halfs connected together or only the left half (with the TRRS disconnected). Also, I noticed the TX LED is lit up all the time on the ProMicro, which is weird to me since I believe TX is for serial communication? And I'm not using serial.

I'd be happy to work on a PR to add a guide for the required configuration for a split keyboard via i2c.

I can flash a non-split firmware on the ProMicro and the 2 sides work fine, so I believe this is a firmware issue. Also, if the problem was on my PCB, I believe the left side should be able to work on its own anyway, without the other half connected?

Thanks

zvecr, 2019-06-24:
For more in depth coverage of split, there is already a work-in-progress PR, #5802, that could help answer some of your questions.

As for an up to date example, lets_split has been updated to split common, and only needs `#define USE_I2C` at the keyboard/keymap level to enforce i2c usage.

As for your immediate issues, i would start with the following:
1. Your row and pin defs are far too small for your defined matrix size, you need 6 row pins and 8 col pins defined.
2. Remove `LIB_SRC += i2c_master.c` from rules.mk as its already included by `SPLIT_KEYBOARD=yes` 
3. Update `#define DEBOUNCING_DELAY 5` to `#define DEBOUNCE 5` as this has changed recently

If you create a draft PR, then some of the above can be made more obvious with githubs code suggestions.

noroadsleft, 2019-06-24:
> 1. Your row and pin defs are far too small for your defined matrix size, you need 6 row pins and 8 col pins defined.

This actually isn't true. Split Common runs its own custom matrix scanning that effectively doubles up the defined pins:

https://github.com/qmk/qmk_firmware/blob/1a2a54c326d0bb0dba899b3098487450d6d9dee6/quantum/split_common/matrix.c#L55

... then the routine uses `ROWS_PER_HAND` for the scanning.

zvecr, 2019-06-24:
@noroadsleft the defined matrix size is 12 x 8, thus my comment on 6 x 8 which is the non doubled size.

noroadsleft, 2019-06-24:
> @noroadsleft the defined matrix size is 12 x 8, thus my comment on 6 x 8 which is the non doubled size is _technically_ still correct.

This made me look and realize I had @LouWii's tree in one tab and the existing Let's Split code in another tab, and I mixed up which one was which. :laughing: 

You are correct, @LouWii's pin definitions here need to be updated.

LouWii, 2019-06-24:
🤦‍♂ how could I miss this!? I'm pretty sure I changed it but I might have mixed up some files. I'll check tonight at home and see what's up with that. 🤞 it's as simple as this!

drashna, 2019-06-24:
can confirm that the issue is that not enough pins are defined for the COL and ROW variables.

LouWii, 2019-06-25:
I fixed the pins definition, now it's working better, thanks guys. I have another problem though, which I'm not sure if it's related to QMK (probably not).

My switches don't work. When I press any of them, nothing gets registered. But, if I manually short a col pin and row pin directly on the ProMicro, the press gets detected. I have no idea what's happening. I flashed back a "non-split" firmware and the switches work fine. No idea why they wouldn't work in split mode.

closed 2019-06-25 by LouWii

zvecr, 2019-06-25:
Glad you got the board working, looking forward to seeing a PR for it.